dss-extensions/OpenDSSDirect.py #2: Cannot load Monitor's byte stream

Muxelmann, 2017-08-24:
An error is thrown when calling the `Monitors.ByteStream()` method:

```
opendssdirect/_lib/construct.py:287: UserWarning: Unsupported dtype 8209 returned for MonitorsV. Please contact developer
```

I got it working in Windows using the `automation` module, e.g.:

```python
def byte_stream():
    """Returns the byte stream of the monitor"""
    arg1 = ctypes.pointer(automation.VARIANT())
    lib.MonitorsV(1, arg1)
    return arg1.contents.value
```

I saw that you mapped `u'variant'` against `ctypes.POINTER(VArg)` in `core.py`, so I'm wondering if I'm not using the library correctly.

---
- Python version: 3.6
- Python architecture: x64
- Operating system and version: macOS
- OpenDSSDirect.py version number 0.2.8

kdheepak, 2017-08-24:
Thanks for bringing this up @Muxelmann, I have been meaning to get to this. You are using the library correctly, this ByteStream function is not implemented yet. I put in a `UserWarning` to say that this functionality was not implemented, but perhaps the message was misleading.

I'm not exactly sure what the `dss.Monitors.ByteStream()` is supposed to return though. This is what I get in Windows when I call that function.

[image]

Do you know what this data represents? Is it supposed to be `int` values? Perhaps @tshort can provide some insight.

kdheepak, 2017-08-24:
I've made some changes in 75ac90b based on [this](https://sourceforge.net/p/electricdss/discussion/861976/thread/337ea669/) thread. I'm still not sure how many rows there are in the bytestream data, that thread uses a `end` keyword. However, in my experiments it seems like the VArg bytestream length is always 272 ( 4 + 4 + 4 + 4 + 256 ). In either case, this will require more looking into.

kdheepak, 2017-08-24:
Made further changes and fixes in dc236e9. This should return the correct values. This makes a function call to `dss.Monitors.Count` to find the number of Monitors.

Muxelmann, 2017-08-25:
When using the `dss.Monitors.ByteStream()` as a `automation.VARIANT()`, I retrieved `monitor_name`, `version`, `size`, `mode`(VI or PQ), number of `data_cols` and the actual `data` table like so:

```python
# Same function as above
def byte_stream():
    """Returns the byte stream of the monitor"""
    arg1 = ctypes.pointer(automation.VARIANT())
    lib.MonitorsV(1, arg1)
    return arg1.contents.value

# Decodes ByteStream
bs = byte_stream()
signature = int.from_bytes(bs[0:3], byteorder='little', signed=True)
if signature != 43756:
    print('Monitor byte stream\' signature is bad')
    return None
else:
    monitor_name = name()
    version = int.from_bytes(bs[4:7], byteorder='little', signed=True)
    size = int.from_bytes(bs[8:11], byteorder='little', signed=True)
    mode = int.from_bytes(bs[12:15], byteorder='little', signed=True)
    header = bytes(bs[16:271]).decode('ascii').rstrip('\0')
    data_cols = header.count(',') + 1
    # Getting data is a bit challenging
    data = []
    for i in range(272, len(bs), 4):
        byte_str = bytes(bs[i:i + 4])
        val = struct.unpack('<f', byte_str)[0]
        data.append(val)
    data = np.array(data).reshape([-1, data_cols])
```

If we can find a way of implementing this without `automation.VARIANT()`, then it would work on Linux and Debian, too! I could probably have written the data extraction in a single line (I didn't test this code):

```python
# Getting data is a bit challenging
data = [struct.unpack('<f', bytes(bs[i:i + 4]))[0] for i in range(272, len(bs), 4)]
data = np.array(data).reshape([-1, data_cols])

# INSTEAD OF
data = []
 for i in range(272, len(bs), 4):
    byte_str = bytes(bs[i:i + 4])
    val = struct.unpack('<f', byte_str)[0]
    data.append(val)
data = np.array(data).reshape([-1, data_cols])
```

kdheepak, 2017-08-25:
Thanks for that code block! I've updated the code to use this. Question: Can you check if you can get a working example of a Monitor using the test case that I have in the repository? I'm unable to get data from the ByteStream if I add a Monitor to the test system (perhaps I'm doing something incorrectly).

```python
# Read file
dss.run_command('Redirect /path/to/repo/tests/data/13Bus/IEEE13Nodeckt.dss')
# Add monitor
dss.run_command('New Monitor.monitor_name element=Transformer.Sub')
# Solve
dss.run_command('Solve')

# Read ByteStream
dss.Monitors.ByteStream()
# Returns empty pd.DataFrame
```

I'd like to add a test for this.

kdheepak, 2017-08-25:
I'm not able to use the code block that you shared as is, since that will not work on non Windows. But I think the current implementation should work across all operating systems. When you get the chance, if you can share how you are using the Monitors or test the ByteStream function on your code that would be great.

kdheepak, 2017-08-25:
I used Test/PVSystemTest.dss from the OpenDSS repo and was able to get this to work. @Muxelmann Thanks for bringing this up! This would work now.

Muxelmann, 2017-08-26:
Thank you @kdheepak, I do no longer get an error and the `pd.DataFrame` is a good implementation. However, I do not seem to get the right values on macOS. I changed your proposed test case (IEEE 13 bus) slightly and am running the following code:

```python
import opendssdirect as dss
import pandas as pd

# Read file
dss.run_command('compile OpenDSSDirect.py/tests/data/13Bus/IEEE13Nodeckt.dss')
# Add monitor
dss.run_command('new Monitor.load_vi Element=Load.671 Terminal=1 Mode=0')
dss.run_command('new Monitor.load_pq Element=Load.671 Terminal=1 Mode=1')
# Set default load shape
idx = dss.Loads.First()
while idx > 0:
    dss.Loads.Daily('default')
    idx = dss.Loads.Next()
# Set solution to daily
dss.Solution.Mode(1)
# Solve
dss.Solution.Solve()

# Read ByteStream
assert dss.Monitors.Count() > 0
idx = dss.Monitors.First()
while idx > 0:
    dss.Monitors.Show()
    data = dss.Monitors.ByteStream()
    print(dss.Monitors.Name())
    print(data.head())
    print('')
    idx = dss.Monitors.Next()
# Returns `pd.DataFrame` with misaligned data
```

The console output is:

```
load_vi
         hour      t(sec)          V1    VAngle1    V2  VAngle2           V3  \
0  180.398804 -152.830917  180.448181  87.160004  15.0      0.0  2380.259033   
1  182.317719 -152.857758  182.369232  87.133095  16.0      0.0  2378.379150   
2  184.130081 -152.885071  184.183868  87.105705  17.0      0.0  2378.175293   
3  184.327377 -152.888077  184.381424  87.102692  18.0      0.0  2386.726074   
4  176.080795 -152.765823  176.125214  87.225281  19.0      0.0  2391.189209   

    VAngle3           I1     IAngle1           I2     IAngle2          I3  \
0 -4.312458  2492.867676 -121.965225  2349.177246  116.855621  182.318253   
1 -4.355270  2492.656738 -121.981331  2346.643555  116.830704  184.131302   
2 -4.359942  2492.634277 -121.983055  2346.365723  116.827965  184.328674   
3 -4.165679  2493.607910 -121.909386  2357.860596  116.941010  176.079346   
4 -4.063947  2494.107422 -121.871147  2363.890137  117.000404  171.771103   

     IAngle3  
0 -32.876347  
1 -32.904171  
2 -32.907242  
3 -32.782791  
4 -32.715878  

load_pq
           hour        t(sec)      S1 (kVA)  Ang1      S2 (kVA)          Ang2  \
0  0.000000e+00  0.000000e+00  0.000000e+00   0.0  0.000000e+00  0.000000e+00   
1  4.971199e-34  1.401298e-45  9.184110e-41   0.0  5.327102e-34  1.401298e-45   
2  0.000000e+00  0.000000e+00  0.000000e+00   0.0  0.000000e+00  0.000000e+00   
3  0.000000e+00  0.000000e+00  0.000000e+00   0.0  0.000000e+00  0.000000e+00   
4  0.000000e+00  0.000000e+00  0.000000e+00   0.0  0.000000e+00  0.000000e+00   

       S3 (kVA)          Ang3  
0  1.524613e-42  0.000000e+00  
1  5.412766e-34  1.401298e-45  
2  0.000000e+00  0.000000e+00  
3  0.000000e+00  0.000000e+00  
4  0.000000e+00  0.000000e+00  
```

But the real data is different (see below). The `ByteStream()` appears to return a subset that start somewhere in the middle of the complete dataset. Maybe a pointer isn't set correctly?

[image]
[image]

Muxelmann, 2017-08-26:
Actually, when changing the last pointer shift on line 315 in `opendssdirect/_lib/construct.py` from:

```python
a_ptr.value = a_ptr.value + 256 * ctypes.sizeof(p._type_)
```

to

```python
a_ptr.value = a_ptr.value + 256
```

Then the data aligns correctly:

```
load_vi
   hour  t(sec)           V1   VAngle1           V2     VAngle2           V3  \
0   1.0     0.0  2426.968262 -2.944626  2500.498535 -121.417542  2400.753906   
1   2.0     0.0  2421.092041 -2.738686  2501.857910 -121.338608  2414.290039   
2   3.0     0.0  2424.359863 -2.663408  2502.232666 -121.310020  2418.722656   
3   4.0     0.0  2429.338379 -2.544719  2487.079834 -121.276520  2425.445068   
4   5.0     0.0  2430.071777 -2.541564  2487.291504 -121.264038  2410.141602   

      VAngle3          I1    IAngle1          I2     IAngle2          I3  \
0  117.654610  122.895622 -31.976622  122.892387 -151.967377  122.911186   
1  117.797440  113.420959 -31.834358  113.423813 -151.823395  113.441185   
2  117.841957  110.237038 -31.785839  110.240250 -151.775513  110.255844   
3  117.902046  105.685989 -31.714247  105.691757 -151.708389  105.698235   
4  117.893799  105.347221 -31.712357  105.346909 -151.706055  105.357094   

     IAngle3  
0  88.029305  
1  88.169876  
2  88.217880  
3  88.285980  
4  88.290939  

load_pq
   hour  t(sec)    S1 (kVA)       Ang1    S2 (kVA)       Ang2    S3 (kVA)  \
0   1.0     0.0  298.263763  29.031996  307.292236  30.549829  295.079498   
1   2.0     0.0  274.602570  29.095673  283.770264  30.484787  273.879913   
2   3.0     0.0  267.254242  29.122431  275.846771  30.465498  266.678314   
3   4.0     0.0  256.747040  29.169527  262.863831  30.431866  256.365265   
4   5.0     0.0  256.001312  29.170794  262.028473  30.442024  253.925522   

        Ang3  
0  29.625303  
1  29.627565  
2  29.624083  
3  29.616070  
4  29.602858  
```

kdheepak, 2017-08-26:
Good job narrowing this down. You were right. I don't know what I was thinking when I used `_type_`. And thanks for the PR! I've merged it :)

closed 2017-08-26 by kdheepak

PMeira, 2025-08-08:
I noticed this very old issue is frequently visited, I wonder why... I'll add some notes and edit this post in the future if required due to any updates.

There is little reason to access the monitor's bytestream directly. All Python packages from DSS-Extensions include a function named `AsMatrix` that returns all data from the monitor.

- OpenDSSDirect.py: https://dss-extensions.org/OpenDSSDirect.py/opendssdirect.html#opendssdirect.Monitors.IMonitors.AsMatrix
- DSS-Python: https://dss-extensions.org/DSS-Python/apidocs/dss/dss.IMonitors.html#dss.IMonitors.IMonitors.AsMatrix
- AltDSS-Python: https://dss-extensions.org/AltDSS-Python/apidocs/altdss/altdss.MonitorExtras.html#altdss.MonitorExtras.MonitorObjMixin.AsMatrix

In the upcoming releases, `AsMatrix` works with both the AltDSS (default engine) and EPRI's OpenDSS distribution.

Additionally, OpenDSSDirect.py has `monitor_to_dataframe` in the utils submodule: https://dss-extensions.org/OpenDSSDirect.py/opendssdirect_utils_ref.html#opendssdirect.utils.monitor_to_dataframe

The monitor header (column names) is not included in the bytestream anymore, since several OpenDSS releases ago.