dss-extensions/OpenDSSDirect.py #4: Setting loadshape PMult and QMult

krspetrou, 2017-10-15:
Hello,

I've been having an "issue" with setting up loadshapes with the opendssdirect library. When I was using the COM port, I could set up the PMult and QMult through the COM port, without the need to create a text file.

e.g., dss.Circuit.LoadShapes.Pmult = values

I understand that the PMult and QMult functions in opendssdirect are read-only functions, but is it possible to set these values without having to create a text file each time I want to create a loadshape?

Python version: 3.6.2
Python Arch.: x64
Windows 10
v0.2.9

kdheepak, 2017-10-15:
Good question. Looking at the code right now, it appears that I did not implemented the "set" functions for Pmult and Qmult in the Python package, but it does appear to be a way to do this. I would have to figure out how to map a Python list to the internal data structure that OpenDSS is using. I'll look into this further and report back here if I'm able to figure out how this can be done. 

For now though, you may get away with what you require by doing the following.

```python
import opendssdirect as dss
# ...
# ...
dss.run_command("loadshape.Residential npts=4, interval=1, mult={data}".format(
    data=(0.461402, 0.406596, 0.348176, 0.317331))
)
```

You can run any OpenDSS string in the current active instance using the `run_command` function.

kdheepak, 2017-10-15:
```python
# import opendssdirect
import opendssdirect as dss

# Create a feeder with a "Residential" loadshape here, e.g. ckt5
dss.run_command("Redirect ../OpenDSS-Source/Distrib/EPRITestCircuits/ckt5/Master_ckt5.dss")

# Edit the Residential loadshape values
dss.run_command("loadshape.NewResidential npts=4, interval=1, mult={data}".format(
    data=(0.461402, 0.406596, 0.348176, 0.317331))
)
```

I just tried this on my Mac, and I know it creates a "NewResidential" loadshape, but I'm unable to retrieve any values from it using this interface. Were you able to get values from the loadshape using Pmult or Qmult?

krspetrou, 2017-10-16:
Thanks for the recommendation. It seems to be working just fine like this. After I set up the loadshape I can retrieve the values using Pmult and Qmult.

One more question, which is a little bit unrelated to these. With a storage element, if I want to change the kWrated value while I'm iterating, do I have to do `dss.run_command('edit storage.test kWrated=10')` or is there a way through one of the functions? 

I'm doing very heavy Monte Carlo simulations, where I'm constantly adjusting my storage power  set-points depending on external control, and I'm trying to have my code be as computational efficient as possible. It's to my understanding that the build-in functions are much faster computationally than changing values through the run_command function.

kdheepak, 2017-10-16:
Can you share a simple example (or a minimal working example) of how you are able to set and retrieve the Pmult and Qmult values? I'm curious why it is not working for me.

The Storage class does not have a separate module unfortunately, and unless the next version of OpenDSS supports it there's little I will be able to do about that. It might be worth starting a thread in the OpenDSS forum discussion regarding this, since I've had at least one other person request the same feature. 

You can do what you have suggested though, using `run_command`. I've been using that for my work and it seems to work fine. I figure that the computation bottleneck will probably be the power flow solve anyway. That's not to say we shouldn't use the most efficient alternative in other places. 

I haven't done extensive tests yet, but my initial impression is that both these methods would be able about the same in terms of computational efficiency. The built-in functions map Python objects (int, float, strings) to their corresponding memory addresses and appropriately call functions that exist in the shared library. These functions then probably manipulate a global data structure that has been set up on the start of OpenDSS. The `run_command` instead passes a string to a built-in function, which then parses the string and calls functions that manipulate the same global data structure. If I had to guess I would say that the built-in functions would be faster than using a `run_command` call, but I would be surprised if it were significantly faster. I'm going to caveat this by saying that the above is based on my understanding of how this works and I haven't done any actual tests to prove this one way or the other. If you do decide to test this, I would be curious to hear your results. I'll also comment here if I find anything new to add.

krspetrou, 2017-10-16:
```
dss.run_command('new loadshape.ls npts=4 minterval=1 mult={pdata} qmult={qdata} '.format(
        pdata=(0,1,2,3),
        qdata=(3,2,1,0)))
dss.LoadShape.Name('ls')
print(dss.LoadShape.PMult())
print(dss.LoadShape.QMult())
```
What's important here is that you need to specifically do the `dss.LoadShape.Name('ls')` step, even if `dss.LoadShape.Name()` returns that this loadshape is activated (perhaps a bug?). Otherwise you get a [0.0] return from both PMult() and QMult(). Let me know if you want the full code, but this should probably solve your issue.

On the storage part, when I was using the COM port, I was able to change different properties of storage using `dssObj.ActiveCircuit.CktElements('storage.name').Properties('kWrated').Val = 10`
similarly, `dssObj.ActiveCircuit.CktElements('storage.name).Properties('kWhstored').Val` would return the current value. Any chance this can be implemented with the API without having to reinvent the wheel?

I'm using a very large number of residential storage devices in a large network, where I have to explicitly change the power setpoint at each iteration (1-minute resolution), for very large Monte Carlo analyses. My biggest bottleneck was (with the COM port) communication overhead even with the usage of early binding. Thus why I've moved over to the API,  and I'm trying to assess how much gain I can get in performance from communication overhead reduction.

krspetrou, 2017-10-16:
Actually, disregard my comment about speed. You're right, the speed from setting things through `dss.run_command` is actually better than the build-in functions as you can set multiple things at once, rather than calling multiple functions (i.e., setting the name, then the kW value etc.)

kdheepak, 2017-10-16:
I'm assuming you mean the following, which like you said this is probably faster. Let me know if you do any tests.  

```python
dss.run_command('"""
Storage.storage1 ...
Storage.storage2 ...
""")
```

I'll writing this out so that it's clear for other readers.

kdheepak, 2017-10-16:
I'm assuming you mean the following, which like you said this is probably faster. Let me know if you do any tests.  

```python
dss.run_command("""
Storage.storage1 ...
Storage.storage2 ...
""")
```

I'll writing this out so that it's clear for other readers.

krspetrou, 2017-10-17:
I don't seem to be able to edit multiple elements at once (with only one call of run_command()) like your code, can you please provide your full code for doing so? What I did was test the speed between changing load values using run_command and using functions, as shown below:

```
#test 1
for i in range(10000):
     dss.run_command('edit load.load1 kW={randomValue}'.format(
           randomValue=np.random.randint(1,10))

#test 2
for i in range(10000):
     dss.Loads.Name('load1')
     dss.Loads.kW(np.random.randint(1,10))
```

test 1 resulted in 0.076 seconds while test 2 resulted in 0.12 seconds. Thus I'm (hopefully correctly) concluding that because of less function calls, it's actually more beneficial to change values through run_command. Because with multiple loads/generators/other elements, you have to declare the name first and the value afterwards. The run_command() function probably becomes even more beneficial if you want to change more values (like kVar etc.) as it results in even less calls than using the functions.

Muxelmann, 2017-10-17:
For editing multiple elements at once, have you tried concatenating commands with return line (`\n`) characters to form a long command string and passing that to DSS? E.g.:

```python
cmd_str = ''
for i in range(10000):
    cmd_str += 'edit load.load{loadId} kW={randomValue}\n'.format(
           loadId=i, randomValue=np.random.randint(1,10))
dss.run_command(cmd_str)
```

kdheepak, 2017-10-17:
@Muxelmann thanks for that comment. @krspetrou, the above comment is exactly what I meant you could do. However, I just tested it and `run_command` does not work the way I expected it to. It appears the OpenDSS command to put text strings in it does not support multiple lines, which is a shame. 

I just updated the `run_command` function to split the text string into multiple lines and pass each line individually to the shared library. I'm not sure if this is any better though in terms of performance.

You should be able to run the following now.

```python
data = (0.461402, 0.406596, 0.348176, 0.317331, 0.297966, 0.282850, 0.272295, 0.283828, 0.341328, 0.414316, 0.479789, 0.562966, 0.669150, 0.703979, 0.732955, 0.797333, 0.769829, 0.674118, 0.610082, 0.573815, 0.544508, 0.523452, 0.490362, 0.428271, 0.364141, 0.311291, 0.277558, 0.258140, 0.253313, 0.253260, 0.286763, 0.315469, 0.302003, 0.329289, 0.353486, 0.407727, 0.499632, 0.577457, 0.633720, 0.677271, 0.720847, 0.731836, 0.762074, 0.747741, 0.692251, 0.660239, 0.598773, 0.525279, 0.429261, 0.371024, 0.319406, 0.280964, 0.258888, 0.262047, 0.284129, 0.317343, 0.299316, 0.303677, 0.341022, 0.359032, 0.382728, 0.420038, 0.456558, 0.502796, 0.550967, 0.599999, 0.635393, 0.629176, 0.580221, 0.553666, 0.525043, 0.455468, 0.386889, 0.320885, 0.278678, 0.254492, 0.234632, 0.226806, 0.228450, 0.232186, 0.274818, 0.357099, 0.444837, 0.543418, 0.610842, 0.662514, 0.715087, 0.753087, 0.763470, 0.788505, 0.767802, 0.745549, 0.687159, 0.635900, 0.570774, 0.512385, 0.426915, 0.355955, 0.305904, 0.272631, 0.250143, 0.243873, 0.256337, 0.278248, 0.270563, 0.286751, 0.356775, 0.453317, 0.545758, 0.642501, 0.697484, 0.754513, 0.787503, 0.827447, 0.843194, 0.819344, 0.775522, 0.759322, 0.698191, 0.626271, 0.517058, 0.440146, 0.385810, 0.355596, 0.333255, 0.316371, 0.325334, 0.339413, 0.352520, 0.410332, 0.483962, 0.578218, 0.656703, 0.726933, 0.738130, 0.743080, 0.727616, 0.727622, 0.729137, 0.682610, 0.645382, 0.616571, 0.588006, 0.532728, 0.456847, 0.389629, 0.342083, 0.309152, 0.291590, 0.279426, 0.277311, 0.284265, 0.328004, 0.382145, 0.423179, 0.458197, 0.486030, 0.505006, 0.510994, 0.558051, 0.611396, 0.680335, 0.685179, 0.683729, 0.644327, 0.607171, 0.570981, 0.497545, 0.419785, 0.355048, 0.314897, 0.275360, 0.253850, 0.244096, 0.238327, 0.259979, 0.308657, 0.376682, 0.431188, 0.515431, 0.625970, 0.721770, 0.776194, 0.815224, 0.856860, 0.863248, 0.896504, 0.878647, 0.855263, 0.800273, 0.727522, 0.613530, 0.510110, 0.433298, 0.388327, 0.353745, 0.340545, 0.334115, 0.358525, 0.364241, 0.373258, 0.380978, 0.407108, 0.471562, 0.570014, 0.681148, 0.724752, 0.738713, 0.762262, 0.751336, 0.755120, 0.730816, 0.696989, 0.661265, 0.619659, 0.528532, 0.433934, 0.365550, 0.330603, 0.299605, 0.281171, 0.284771, 0.306665, 0.326507, 0.328157, 0.344994, 0.365084, 0.421600, 0.457383, 0.499826, 0.527713, 0.576358, 0.632612, 0.718134, 0.767207, 0.752692, 0.723143, 0.684985, 0.651335, 0.561269, 0.471103, 0.405741, 0.368826, 0.341299, 0.322187, 0.330379, 0.353003, 0.384284, 0.367353, 0.362685, 0.365184, 0.385681, 0.432178, 0.487839, 0.573910, 0.599392, 0.653539, 0.644328, 0.661872, 0.673151, 0.661318, 0.619222, 0.576279, 0.484887, 0.396754, 0.326218, 0.277234, 0.248440, 0.230141, 0.233088, 0.256295, 0.272814, 0.268265, 0.294248, 0.336172, 0.392558, 0.445296, 0.501948, 0.567498, 0.621639, 0.660740, 0.705947, 0.748478, 0.720155, 0.671855, 0.625216, 0.578276, 0.486219, 0.382811, 0.332076, 0.293346, 0.266161, 0.243342, 0.249123, 0.273839, 0.292757, 0.272849, 0.285673, 0.299345, 0.320089, 0.363782, 0.366463, 0.372940, 0.404592, 0.463447, 0.528461, 0.559465, 0.544903, 0.520665, 0.488452, 0.463324, 0.405346, 0.343821, 0.284547, 0.250379, 0.228432, 0.218290, 0.210222, 0.215025, 0.227389, 0.265271, 0.304013, 0.350410, 0.436952, 0.522704, 0.601119, 0.653073, 0.721352, 0.757029, 0.786925, 0.786519, 0.755909, 0.715329, 0.684631, 0.635847, 0.563708, 0.487144, 0.425808, 0.378179, 0.349561, 0.318257, 0.298426, 0.291413, 0.300995, 0.340138, 0.392493, 0.445461, 0.517954, 0.574104, 0.626023, 0.700390, 0.765751, 0.692952, 0.619647, 0.593581, 0.573238, 0.552788, 0.529775, 0.511872, 0.430711, 0.361147, 0.305952, 0.266638, 0.249554, 0.235663, 0.248658, 0.273327, 0.298538, 0.283177, 0.294354, 0.344358, 0.431642, 0.527819, 0.608143, 0.666156, 0.719861, 0.757836, 0.800185, 0.813521, 0.771273, 0.714651, 0.664005, 0.606847, 0.519168, 0.433427, 0.374943, 0.324055, 0.294030, 0.276603, 0.273869, 0.292592, 0.323166, 0.316819, 0.330361, 0.356527, 0.419903, 0.480001, 0.529775, 0.555829, 0.564952, 0.575206, 0.619264, 0.664706, 0.677536, 0.664211, 0.648093, 0.608155, 0.539393, 0.436716, 0.372999, 0.337197, 0.298261, 0.296063, 0.301567, 0.314774, 0.344464, 0.331917, 0.346998, 0.408022, 0.488193, 0.538103, 0.546777, 0.472040, 0.454767, 0.475658, 0.530907, 0.563255, 0.558434, 0.550431, 0.546135, 0.517023, 0.454260, 0.375456, 0.318092, 0.283534, 0.263638, 0.248015, 0.250597, 0.280941, 0.302298, 0.296310, 0.326849, 0.396112, 0.487521, 0.574947, 0.631074, 0.703861, 0.664223, 0.613347, 0.626094, 0.645388, 0.682256, 0.661323, 0.638705, 0.606593, 0.513475, 0.434023, 0.373281, 0.342307, 0.310595, 0.292273, 0.289722, 0.315675, 0.340769, 0.313507, 0.319907, 0.344947, 0.388480, 0.426432, 0.474350, 0.518402, 0.577334, 0.637810, 0.651977, 0.663822, 0.644375, 0.616529, 0.600417, 0.565364, 0.490379, 0.409318, 0.343061, 0.291389, 0.258653, 0.240266, 0.216009, 0.217707, 0.229982, 0.265607, 0.313342, 0.381172, 0.444071, 0.520759 [... cut ...]
dss.run_command("Redirect ../opendss-mirror/Distrib/EPRITestCircuits/ckt5/Master_ckt5.dss")

r = dss.run_command("""new loadshape.NewResidential1 npts=8760, interval=1, mult={data}
new loadshape.NewResidential2 npts=8760, interval=1, mult={data}
new loadshape.NewResidential3 npts=8760, interval=1, mult={data}
new loadshape.NewResidential4 npts=8760, interval=1, mult={data}
new loadshape.NewResidential5 npts=8760, interval=1, mult={data}
new loadshape.NewResidential6 npts=8760, interval=1, mult={data}
""".format(
    data=data
))
```

tshort, 2017-10-18:
@krspetrou, for your test 2, repeated below, you might want to try using `dss.Loads.Idx(some_num)` instead of `dss.Loads.Name('load1')`. The name lookup is expensive. There's no dict or anything--the Pascal code iterates through the loads, comparing the names one by one. It might even be faster to use a Python dict to map names to indexes. If it's appropriate for your application, iterating through loads is pretty fast.

```py
#test 2
for i in range(10000):
     dss.Loads.Name('load1')
     dss.Loads.kW(np.random.randint(1,10))
```

krspetrou, 2017-10-21:
@Muxelmann Thanks for the input! However, when I first tried it, it wasn't working, but I just saw the message from @kdheepak that this has just been implemented, so I had to update the library for it to work. It works great now, I will run some tests in the near future as I'm working on something more imminent at the moment, but I believe it will be great in terms of performance.

@tshort I've tried to do that although it seems that the library doesn't activate the element when I do it like that. Maybe a bug? To better explain, When I activate the element through the name and I type `some_num=dss.Loads.Idx()`, but when I do `dss.Loads.Idx(some_num)` it doesn't activate the element (I activated some other element first). It returns me 0 when I do that.

krspetrou, 2017-10-21:
One more thing, it seems that the method described by @Muxelmann is very very slow (was running at least a minute before giving up on it) when you have to add a very large number of strings. Better approaches can be found here: https://waymoot.org/home/python_string/

I personally use Method 4, which did the same in less than 3 seconds. Apparently method 6 is the fastest, but I can't do that for various reasons.

kdheepak, 2017-12-14:
I'm closing this issue, let me know if you need anything else!

closed 2017-12-14 by kdheepak