dss-extensions/OpenDSSDirect.py #8: Monitoredobj names for fuses

NicolasGensollen, 2017-12-14:
Hey Dheepak,

I'm working on the SMART-DS networks that have weird name conventions as you probably know. I have a small issue when parsing fuses in DiTTo and the problem seems to be at the opendssdirect level.

More precisely, fuses are modeled in the following way:

```
New Line.Fuse(CRamaEE():S_nCCTT992->S_nCCTT996)_S_2106 bus1=S_nCCTT996.3 bus2=S_nCCTT992->S_nCCTT996_S.3 Length=0.001 Switch=y Units=km Normamps=3000 Emergamps=3000

New Fuse.Fuse(CRamaEE():S_nCCTT992->S_nCCTT996)_S_2106 monitoredobj=Line.Fuse(CRamaEE():S_nCCTT992->S_nCCTT996)_S_2106 enabled=no
```

So, you have a dummy line object and a fuse attached to it. 
When I am doing the following:

```python
dss.utils.class_to_dataframe('fuse')
```

I get the following output:

```
...,
'fuse.fuse(cramaee():s_ncctt992->s_ncctt996)_s_2106': {u'Action': u'',
  u'Delay': u'0',
  u'FuseCurve': u'Tlink',
  u'MonitoredObj': (u'Line.FuseCRamaEE:S_nCCTT992->S_nCCTT996_S_2106',),
  u'MonitoredTerm': u'1',
  u'RatedCurrent': u'1.0',
  u'SwitchedObj': u'',
  u'SwitchedTerm': u'1',
  u'basefreq': u'60',
  u'enabled': False,
  u'like': u''}
```

It looks like all the parentheses were removed from the monitoredObj which appears as "Line.FuseCRamaEE:S_nCCTT992->S_nCCTT996_S_2106" instead of "Line.Fuse(CRamaEE():S_nCCTT992->S_nCCTT996)_S_2106".

Do you see any reason why it behaves like that?

Thanks!
Nicolas

kdheepak, 2017-12-14:
Yes. That's probably my doing. I clean the output from opendss to get a python object, seen in this function -> https://github.com/NREL/OpenDSSDirect.py/blob/56cd649/opendssdirect/utils.py#L103-L121. I'll have to look into it exactly to see how to not do it for your specific use case.

NicolasGensollen, 2017-12-14:
Yep, it looks like this is the problem. 
Sorry for that, these names are crazy...

kdheepak, 2017-12-18:
Can you try this in the latest `master` branch? It should be fixed. If it is not please reopen this issue.

closed 2017-12-18 by kdheepak

kdheepak, 2017-12-18:
I made a couple of other changes to the `_evaluate_expression` function. This will ensure that only strings that start with "(" and end with ")" get converted to a tuple. However, this still could result in some false positives. So to bypass this, I've also added the option for a user definable function `transform_string` that allows more customization.

commit cda20d6a (2017-12-18): Update utils.py to not use tuple for objects

Fixes #8