dss-extensions/OpenDSSDirect.py #10: negative mutual resistances

NicolasGensollen, 2018-01-10:
Hey Dheepak,

I received some comments from CYME for the SMART-DS datasets saying that some lines have negative mutual resistances and that they do not support that.
I had a look at that and I noticed something weird when reading the OpenDSS input with OpenDSSdirect.

For example, the following line:

```bash
New Line.CLineaCable(CRamaEE():D2CLV32500->D2_nCCTT7324_BT) bus1=D2_nCCTT7324_BT.1.2 bus2=D2CLV32500.1.2 Linecode=1P_UG_Sweetbriar_TRPLX4/0 Length=0.0322386 Switch=n Repair=60 Units=km
```

which has the following linecode:

```bash
New LineCode.1P_UG_Sweetbriar_TRPLX4/0 nphases=2 r1=0.331365 x1=0.0869423 c1=1.11111e-07 Units=km basefreq=60 normamps=290 faultrate=0.1
```

becomes the following object in OpenDSSdirect:

```python
lines=dss.utils.class_to_dataframe('line')
print(lines['line.clineacable(cramaee():d2clv32500->d2_ncctt7324_bt)'])

{u'B0': u'0.6031858',
 u'B1': u'4.188786E-8',
 u'C0': u'1.6',
 u'C1': u'1.11111E-7',
 u'EarthModel': u'Deri',
 u'Rg': u'0.01805',
 u'Switch': False,
 u'Xg': u'0.155081',
 u'basefreq': u'60',
 u'bus1': u'd2_ncctt7324_bt.1.2',
 u'bus2': u'd2clv32500.1.2',
 u'cmatrix': [u'0.5333334 |0.5333333 0.5333334'],
 u'cncables': u'',
 u'emergamps': u'600',
 u'enabled': True,
 u'faultrate': u'0.10000000000000001',
 u'geometry': u'',
 u'length': u'0.0322386',
 u'like': u'',
 u'linecode': u'1P_UG_Sweetbriar_TRPLX4/0',
 u'normamps': u'290',
 u'pctperm': u'20',
 u'phases': u'2',
 u'r0': u'0.1784',
 u'r1': u'0.331365',
 u'repair': u'60',
 u'rho': u'100',
 u'rmatrix': [u'0.2803767 |-0.05098833 0.2803767'],
 u'spacing': u'',
 u'tscables': u'',
 u'units': u'km',
 u'wires': u'',
 u'x0': u'0.4047',
 u'x1': u'0.0869423',
 u'xmatrix': [u'0.1928615 |0.1059192 0.1928615']}
```

The rmatrix is  [u'0.2803767 |-0.05098833 0.2803767']. That is, the mutual resistance between the 2 phases is negative. 

My understanding is that this value is calculated using 1/3*(R0-R1), which gives this negative value. Since R0 is not given in the linecode, I don't understand where it comes from.
Anyway, this looks strange to me and after talking to Tarek about this, he doesn't have an explanation either.
Are we missing something here?

Thanks a lot!

kdheepak, 2018-01-10:
[This function](https://sourceforge.net/p/electricdss/code/2092/tree/trunk/Source/General/LineCode.pas#l248) is being called to get the `rmatrix` values. That `Z` used in that function is [created in this function](https://sourceforge.net/p/electricdss/code/2092/tree/trunk/Source/General/LineCode.pas#l602) and from what I can tell that looks a little different from what you have posted above. This question might be better answered by someone that is more familiar with this code, you can probably get a more detailed answer if you posted it on the [sourceforge forum](https://sourceforge.net/p/electricdss/discussion/).

NicolasGensollen, 2018-01-10:
Thanks for the quick reply and the links. 
Looking at the function 

```bash
Procedure TLineCodeObj.CalcMatricesFromZ1Z0;
```

it looks like it is computing the impedance matrix in the following way:

```
              2*Z1+Z0    Z0-Z1      Z0-Z1
Zapprox=1/3*  Z0-Z1      2*Z1+Z0    Z0-Z1
              Z0-Z1      Z0-Z1      2*Z1+Z0
```
which is the usual approximation when you only have the sequence impedances. 
But the thing I don't get is the value of Z0 (Z0=0.1784+j0.4047) used (Z1 is the value provided in the linecode). 
I don't know if it is a default value or if it is computed somehow using the provided info in the lincode.

kdheepak, 2018-01-10:
Yes, that is the default value for [R0](https://sourceforge.net/p/electricdss/code/2092/tree/trunk/Source/General/LineCode.pas#l551) and [X0](https://sourceforge.net/p/electricdss/code/2092/tree/trunk/Source/General/LineCode.pas#l552).

NicolasGensollen, 2018-01-10:
OK, nice find! 
I'm gonna talk to Tarek about this and we'll see with CYME what we can do about it.
Thanks a lot!

kdheepak, 2018-01-10:
Great! Let me know if you need more information.

closed 2018-01-10 by kdheepak