dss-extensions/OpenDSSDirect.py #11: .sng and .dbl file formats don't work

kdayday, 2018-04-23:
Working with Python 2.7 on Centos 6 with OpenDSSDirect.py version 0.2.14, loadshapes using OpenDSS .sng and .dbl file formats are not loaded in correctly. The first few values are very tiny numbers (10^-310), and after that all the values are 0. Using a .csv file format works correctly on Centos 6.  Using the .sng and .dbl files (as well as .csv's) works on a Windows 7 platform.

PMeira, 2018-08-13:
I tested this on the `dss_capi_cffi` branch and found no issues with openSUSE and Ubuntu, so it should work after the next OpenDSSDirect.py release.

Since the branch uses the backend from `dss_python`, which is now built using CentOS 5, it should work on CentOS 6 too.

kdheepak, 2018-08-13:
@kdayday Let us know if you are still experiencing this issue.

closed 2018-08-13 by kdheepak