dss-extensions/OpenDSSDirect.py #12: Monitors_to_dataframe() NaN's

jdlara-berkeley, 2018-06-02:
Please provide the following information before submitting your issue. Thank you!

- Python 3.6
- Architecture x64
- MaxOSX
-  OpenDSS direct 0.2.14

The command ```dss.utils.monitors_to_dataframe()``` return a DataFrame with NaNs. 

The monitors are correct since running the export command return valid CSV files.

kdheepak, 2018-06-02:
Thanks for reporting, I'll look into this.

kdheepak, 2018-08-13:
Can you check if you are still experiencing this in v0.3.0?

PMeira, 2018-08-17:
I ran the test case from [123Bus/SolarRamp.DSS](https://github.com/PMeira/electricdss-tst/blob/master/Distrib/IEEETestCases/123Bus/SolarRamp.DSS) and found a little issue -- `get_float64_array` was used but not imported in `dss.Monitors`. Since it was a very simple fix, I tested locally and committed it to master.

After I added the missing import, the results seem OK. The byte-stream or the channels themselves are not included in the current version. Maybe the byte-stream was being misinterpreted before, resulting in NaNs.

We could add a similar function, e.g. `monitor_to_dataframe`, with returns a more complete dataframe for a single monitor, including all the channels with a proper header. I'll make a PR about this soon since I could see myself using it.

kdheepak, 2018-08-17:
Ah sorry about that. Thanks for fixing it.

kdheepak, 2018-08-17:
There must have not been a test case for that, we should add that as well.

commit 42c00f9b (2021-01-04): Fix Monitors.AsMatrix, add utils.monitor_to_dataframe and related test.

Closes #12
commit d341e8eb (2021-01-04): Fix Monitors.AsMatrix, add utils.monitor_to_dataframe and related test.

Closes #12
PR #92 (2021-01-04): Implement utils.monitor_to_dataframe
Small changes to properly add `monitor_to_dataframe`, closes #12.
commit 3b553d13 (2021-01-04): Fix Monitors.AsMatrix, add utils.monitor_to_dataframe and related test.

Closes #12
closed 2021-01-04 by PMeira