dss-extensions/OpenDSSDirect.py #14: Using older version of OpenDSS

NicolasGensollen, 2018-06-18:
Hey @kdheepak,

I was wondering if it is possible to use older versions of OpenDSS through OpenDSSdirect. I believe the current version used is 8.0 (I'm using OpenDSSdirect v.0.2.14):

```python
In [2]: import opendssdirect as dss

In [3]: dss.Basic.Version()
Out[3]: u'Version Version 8.0 release 1 build 0\n (64-bit build); License Status: Open '
```

This could be important since we are porting some QSTS code from Matlab to Python and it seems like there is some issue with version 8 and we should use version 7 instead (don't know exactly what is the problem but I can ask Barry for more details if needed...).

kdheepak, 2018-06-18:
I'll look into this, but it may not be straightforward.

kdheepak, 2018-06-20:
If there is an issue with version 8, we should report to original developers. If they are unable to fix, then we can consider a method to revert to an older version if required. I'm closing this issue for now. Feel free to reopen when either the issue with version 8 / the need for version 7 is identified.

closed 2018-06-20 by kdheepak