dss-extensions/OpenDSSDirect.py #17: Control Queues

NicolasGensollen, 2018-07-19:
Hi @kdheepak,

I'm trying to port to Python some Matlab code using control queues. 
For example:

```matlab
CQ = circuit.CtrlQueue.QueueSize; %returns the number of items in control queue
```

So far I couldn't find a way to do the same thing with ```OpenDSSDirect```. Am I missing something?

Thanks!

tshort, 2018-07-19:
Maybe `CtrlQueue.NumActions()`?

NicolasGensollen, 2018-07-27:
Hi @tshort, I had a look at ```OpenDSSDirect.jl``` to see if control queues were working and, if yes, try to understand why it's not the case for ```OpenDSSDirect.py```.
I haven't used Julia too much in the past, so this might just be me doing something wrong, but when I try using ```CtrlQueue``` I get the following error message:

```Julia
julia>CtrlQueue.Show()

ccall: could not find function CtrlQueueI in library /Users/ngensoll/.julia/v0.6/OpenDSSDirect/src/../deps/apple/libopendssdirect.dylib

Stacktrace:
 [1] Show() at ./<missing>:0
```

```Julia
julia>CtrlQueue.NumActions()

ccall: could not find function CtrlQueueI in library /Users/ngensoll/.julia/v0.6/OpenDSSDirect/src/../deps/apple/libopendssdirect.dylib

Stacktrace:
 [1] NumActions() at ./<missing>:0
```

Any idea what might be causing this?

PMeira, 2018-08-13:
I looked into this and in the previous version of OpenDSSDirect.py, due to issues in the DDLL, CtrlQueue would not work at all. Looking on the code of the official OpenDSS Direct DLL (both v7 and v8):

- On Windows, there is [an uninitialized variable](https://sourceforge.net/p/electricdss/code/2254/tree/trunk/Source/DDLL/DCtrlQueue.pas#l93) that could possibly crash (access violation) the whole program.
- On Linux/Mac (or with the DLL built with Free Pascal in general), the DDLL does not include the CtrlQueue interface file ([DCtrlQueue is commented](https://sourceforge.net/p/electricdss/code/2254/tree/trunk/Source/DDLL/OpenDSSDirect.lpr)). This explains the message `could not find function CtrlQueueI in library`.

With OpenDSSDirect.py v0.3.0, which uses DSS C-API instead, both of the issues should be fixed. The API is fully exposed in the [CtrlQueue submodule](https://github.com/NREL/OpenDSSDirect.py/blob/87b9dfab5ab8be32f057ac5a562ef55a53eb4418/opendssdirect/CtrlQueue.py) (based on [CAPI_CtrlQueue.pas](https://github.com/PMeira/dss_capi/blob/12774f47240b3fc1a97c7be52001faaafebc4d45/src/v7/CAPI_CtrlQueue.pas)). 

The code may still need some tweaks are Pascal level, so please let me know if other issues still remain present.

NicolasGensollen, 2018-08-13:
Thank you guys, the latest version of ```OpenDSSdirect.py``` seems to solve a lot of issues I had. 👍  
I had a brief look but didn't test the control queue results. I'll try to spend some time on this during the week. I'm closing this and will reopen or open a new issue if I spot other related bugs.

closed 2018-08-13 by NicolasGensollen