dss-extensions/OpenDSSDirect.py #24: Update tests to be more thorough

kdheepak, 2018-08-15:
**I'm submitting a ...**
- [x] feature request

### Feature Request

Update tests so that they are more complete

e.g. some tests need params

```
assert dss.Bus.GetUniqueNodeNumber()
assert dss.Circuit.Capacity()
assert dss.Circuit.SetActiveBusi()
assert dss.CktElement.Close()
assert dss.CktElement.IsOpen()
assert dss.CktElement.Open()
assert dss.CktElement.Variablei()
```

PMeira, 2018-08-17:
Monitors also need more tests. Probably adding one or two monitors via `run_command()` in https://github.com/NREL/OpenDSSDirect.py/blob/30a2060ef1b6986be3a80a49a5b474951d994109/tests/test_opendssdirect.py#L1965 and validating the results is enough.

kdheepak, 2018-08-17:
Note for future reference: See issue #29 and PR #30 to identify many functions that are not currently tested.

PMeira, 2022-07-16:
The new modules `Parallel` and `Storages` are missing tests now, as well as some recent other new functions in other modules.