dss-extensions/OpenDSSDirect.py #25: Bug with `dss.Lines.R()` and `dss.Lines.X()`

kdheepak, 2018-08-15:
Is there a bug with `dss.Lines.R()` and `dss.Lines.X()`? If so, what is it? Are there steps to fix it that can be outlined?

PMeira, 2018-08-15:
Do you mean `dss.Lines.RMatrix()` and `dss.Lines.XMatrix()`? Or all line parameters?

The scaling changed in [OpenDSS r1919 from 2017-05-26](https://sourceforge.net/p/electricdss/code/1919/) and was [partially incomplete](https://sourceforge.net/p/electricdss/discussion/experts/thread/459afcb4/) for some attributes. In the current official OpenDSS, `DSS.ActiveCircuit.Lines.R1` and `DSS.ActiveCircuit.Lines.X1` are still using the old scaling, while the rest of them were changed in r1919. In ODDpy, all values use the same scaling since I patched the code.

kdheepak, 2018-08-15:
I meant to ask about `R1()` and `X1()`, I've changed the title to reflect it. These tests were commented out in the tests folder.

```python
#    assert dss.Lines.R1() == 0.058 # DISABLED (BUG IN OPENDSS)
#    assert dss.Lines.X1() == 0.1206 # DISABLED (BUG IN OPENDSS)
```

kdheepak, 2018-08-15:
So this doesn't appear to be a bug then. Thanks for documenting it over here. I'll update the tests to use the new values.

PR #27 (2018-08-15): Update Python tests
Fixes #25
closed 2018-08-15 by kdheepak