dss-extensions/OpenDSSDirect.py #31: dss.Basic.DataPath persists across process

kdheepak, 2018-08-20:
**I'm submitting a ...**
- bug

- OpenDSSDirect.py version number: v0.3.4

	```python
	import opendssdirect as dss
	print(dss.__version__)
	```

### Bug

* **What is the current behavior?**

Calling dss.Basic.DataPath with an argument results in the change persisting across processes.

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**

It should reset to some default value.

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.**

[image]

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

Potentially related to https://github.com/PMeira/electricdss-src/issues/3

kdheepak, 2018-08-20:
Updated to the latest version of `dss_python` and this issue went away. This issue https://github.com/PMeira/electricdss-src/issues/3 must have resolved it, I just had to update to the latest version. I've made `dss_python>=0.9.8` in 81436a56b88.

closed 2018-08-20 by kdheepak

PMeira, 2018-08-20:
The changeset for this was mostly https://github.com/PMeira/electricdss-src/commit/1528842ef9758ef8b334c2d66ca800c4c689007c -- the fix for https://github.com/PMeira/electricdss-src/issues/3 is not released yet, should be out this week though.

kdheepak, 2018-08-20:
Got it. Thanks!