dss-extensions/OpenDSSDirect.py #33: Setting ActiveClass e.g. `dss.Basic.SetActiveClass(b"Load")` not working as expected

kdheepak, 2018-08-20:
### Versions

- [ ] Python version: 3.6
- [ ] Python architecture: x64
- [ ] Operating system and version: MacOS
- [ ] OpenDSSDirect.py version number: v0.3.4

### Bug

* **What is the current behavior?**

```python
dss.Basic.SetActiveClass("Load")
dss.ActiveClass.Name()  # Returns "source"
```

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**

```python
dss.Basic.SetActiveClass("Load")
assert dss.ActiveClass.Name() == "Load"
```

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.**

Run tests after changing line 304 in [tests/test_opendssdirect.py](https://github.com/NREL/OpenDSSDirect.py/blob/e89a636/tests/test_opendssdirect.py#L304)

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

Updated to the latest version of `dss_python==0.9.8`

PMeira, 2018-08-20:
`dss.ActiveClass.Name()` returns the name of the current object of the active class. `dss.ActiveClass.ActiveClassName()` returns the actual class name, so this should work:

```
assert dss.ActiveClass.ActiveClassName() == "Load"
```

kdheepak, 2018-08-20:
Of course. My bad, I fixed this in 8c0d5c4ed10235d26513bb93bb3fb28858530296.

closed 2018-08-20 by kdheepak