dss-extensions/OpenDSSDirect.py #35: Circuit Total Power Mismatch

AkshayKuJain, 2018-08-20:
### Versions

- [ ] Python version: 3.6 
- [ ] Python architecture: x64
- [ ] Operating system and version: Windows 10
- [ ] OpenDSSDirect.py version number: 0.3.4

### Bug

* **What is the current behavior?**
I am simulating a distribution feeder model, where the normalized substation load profile is applied to all the feeder loads. The circuit is then solved at each hour of the year and the total circuit power is recorded using dss.Circuit.TotalPower(). The total circuit power does not match closely with the substation load profile when OpenDSS Direct version 0.3.4 is used. However, they match closely using version 0.2.14 and also using the COM interface with Matlab where I am using OpenDSS version  7.6.5.64 (64-bit build). I have attached the loadshape plots, and a comparison of some of the results. Unfortunately I can't upload the DSS files.

[DSS_comparisons.pdf](https://github.com/NREL/OpenDSSDirect.py/files/2303957/DSS_comparisons.pdf)

PMeira, 2018-08-21:
Thank you for reporting.  We should investigate this but I can take some days depending on what we find.

Some points to consider:

- Most of the results are [validated with COM](https://github.com/PMeira/dss_python#testing) using the DSS samples in the official distribution. `Circuit.TotalPower` is included in the comparison.
- On Windows, ODDpy 0.2.14 used the official DLL that was built with Delphi and it's marked as 7.6.5.52. Some deviation should then be expected, but if it's large we should of course investigate and fix it. From my experience, one potential issue from I could imagine is the implementation of the matrix inverse.
- We accumulated many small changes/fixes since 7.6.5.64 was released, though in a glance I couldn't catch something directly related: https://github.com/PMeira/electricdss-src/compare/84affcdc4b094766f2f443f1f510cbfb237577b9...master
- For ODDpy 0.3+, KLUSolve is built from the upstream SuiteSparse source code (including KLU) plus the [original KLUSolve code](https://github.com/PMeira/dss_capi/tree/b3e791b8920bb118d71053cf6825fe8cbe981052/klusolve). Since KLU is the linear solver behind OpenDSS, this could introduce some potential difference. There are some very minor changes in the new SuiteSparse we use.

First we need to:
- [ ] Create a test case to reproduce the issue

With that done, I could imagine several things to try with it:

- [ ] Compare OpenDSS v7 vs. the latest v8
- [ ] Replace the KLUSolve in dss_python with the DLL from the official distribution
- [ ] Test older dss_python versions (won't be compatible with ODDpy) -- if an error was introduced recently, this would be the easiest way to find it.
- [ ] Build our Windows OpenDSS DLL with Delphi, compare compilation flags, etc.
- [ ] Check the [matrix inverse](https://github.com/PMeira/electricdss-src/blob/13b9e54fd595a9622f25ba1753cae06687519fd3/Source/Shared/Ucmatrix.pas#L140), both in general and Delphi vs. FPC. If we find any issue, it would be better to use the one provided in LAPACK when available (e.g. in Python) since most LAPACK implementations are extensively validated.

kdheepak, 2018-08-21:
I'll try to generate a test case that I can share that reproduces the issue in the parent comment.

PMeira, 2018-08-21:
@kdheepak Thanks, I can take care of the rest on Windows if you prefer.

One more thing we could do:
- [ ] Compare the output on Windows vs Linux (or MacOS) for the same system using ODDpy 0.2.14. Since Linux and MacOs already used Free Pascal before, it would be an easier test.

kdheepak, 2018-08-23:
I ran a very simple test and can say that the problem is not actually with the `dss.Circuit.TotalPower()` function.

[image]

kdheepak, 2018-08-23:
OpenDSS v816_01 gives results are do not match the v0.3.4 results, when comparing results from the GUI to that from Python.

kdheepak, 2018-08-23:
v765 and v816_01 both give the same results, which are not the same as v0.3.4. Tested using x64 bit for both versions.

kdheepak, 2018-08-23:
I replaced the KLUSolve from the official distribution in dss_python and got the same result that we get in v0.3.4.

kdheepak, 2018-08-23:
We've narrowed it down to running `allocateloads`. Without that command in the OpenDSS master file, the total circuit power is approximately equal (to 2 decimal places) for the real and imaginary components. Once `allocateloads` is run, the difference becomes much larger.

PMeira, 2018-08-23:
@kdheepak Well done! Your results so far suggest it's a Delphi vs. Free Pascal issue then, since on Mac the results didn't change across ODDpy versions.

Looking into the test files from the official distribution, `allocateloads` are not present in any of them. No wonder we didn't catch this before. It doesn't seem we use it in our lab at Unicamp either.

I did a simple compilation to test if it would be possible to build the DLL with Delphi, at least for testing, and it seems it wouldn't be too much trouble (https://github.com/PMeira/electricdss-src/issues/4).

kdheepak, 2018-08-24:
This is really odd. I went through the files that I have line by line, and the only things that are in those files that are not in the 13 node test system are EnergyMeters and Monitors. Additionally, I've added an EnergyMeter and Monitor to the 13 node test system in the `tests/data` folder in this repository, and when I run `dss.run_command("Dump AllocationFactors")` it returns an empty file. However, when I read the source files I have that I'm not able to share, and run `dss.run_command("Dump AllocationFactors")` I get a long list of loads and their allocationfactors. Any idea what is going on here?

These are the lines I added to the `IEEE13Nodeckt.dss`

```
! add to bottom of IEEE13Nodeckt.dss
New EnergyMeter.Energy_Meter element=Line.650632 terminal=1
New monitor.sub1 element=Transformer.Sub terminal=2 mode=0
New monitor.sub2 element=Transformer.Sub terminal=1 mode=65 PPolar=No

allocateloads

Solve

```

kdheepak, 2018-08-24:
On a somewhat unrelated note, `Dump AllocationFactors` tries to write a file into `/tmp/OpenDSSCMD/AllocationFactors.txt` instead of the current folder.

kdheepak, 2018-08-24:
I also tried changing the power factors on the loads and that did not make a difference to this result. Oddly, when I change the power factor the total power is still the same. Like the power factor of the load is not being taken into effect?

[image]

kdheepak, 2018-08-24:
Thanks for the detailed answers! Sorry, I keep forgetting that the fix for the working directory is not in a release yet. And that power factor explanation makes sense, thanks.

The linked PDF is interesting; I hadn't seen that before. I was trying to modify the IEEE13 node feeder to replicate the issue. But it looks like I won't be able to get to this until later in the weekend unfortunately.

PMeira, 2020-08-02:
I imagine this is stale and we can close it. Feel free to reopen.

closed 2020-08-02 by PMeira