dss-extensions/OpenDSSDirect.py #36: OpenDSS version is returned as 'Version Unknown'.

activeshadow, 2018-08-23:
### Versions

- [x] Python version: 3.6.5
- [x] Python architecture: x86_64
- [x] Operating system and version: Debian 9
- [x] OpenDSSDirect.py version number: 0.3.4

### Support Question

I need to figure out what version of the underlying OpenDSS library this package is using. Calling `opendssdirect.Basic.Version()` returns `Version Unknown. (64-bit build); License Status: Open`.

Perhaps this is related to #32?

kdheepak, 2018-08-23:
Yes you are right it's related to #32 but we didn't explicitly discuss it there. Thanks for opening the issue. I'll leave this open until we address that issue.

activeshadow, 2018-08-24:
Thanks @kdheepak! Any chance you can go ahead and tell me now what version of the core OpenDSS library is being used?

kdheepak, 2018-08-24:
Yes sure, OpenDSSDirect.py is using dss_python which binds to a version of OpenDSS that is based on OpenDSS revision 2246. @PMeira has added many fixes from that version, some of which are listed [here](https://sourceforge.net/p/electricdss/discussion/experts/thread/6017681a/). If you are interested in having other people reproduce your builds, then they would just need OpenDSSDirect.py v0.3.4 and dss_python v0.9.8. If writing up the documentation or for a paper describing the version you used, I would say something like "OpenDSSDirect.py v0.3.4 with dss_python v0.9.8 which interfaces with OpenDSS v7 r2246m". The "m" in "OpenDSS v7 r2246m" signals that it has been modified.

kdheepak, 2018-11-19:
You can now use the following to get the version of OpenDSS being used

```python
>>> import opendssdirect as dss
>>> dss._utils.api_util.get_string(dss.dss_lib.DSS_Get_Version())
'DSS C-API Library version  revision 4f258c0c5f23c798d27390e47a9a9fc7e8f8a86f based on OpenDSS SVN 2395 (v7/classic variation) (64-bit build); License Status: Open '
```

I might add a utility function in `opendssdirect` so that this is easier to access.

PMeira, 2018-11-19:
@kdheepak  `opendssdirect.Basic.Version()` returns that now too

kdheepak, 2018-11-19:
Yes, of course, I forgot about that function.