dss-extensions/OpenDSSDirect.py #37: Documentation request: Include example Jupyter notebook for load flow or transient simulation

sibyjackgrove, 2018-10-08:
### Versions

- [x] Python version: 3.6
- [x] Python architecture: x64
- [x] Operating system and version: Windows 10
- [x] OpenDSSDirect.py version number: v0.3.4

### Feature Request
 Please include an example Jupyter notebook that shows how to do a load flow or transient simulation for a simple test system. It would be very helpful for new users to easily start using the package and understand the API.

kdheepak, 2018-10-08:
Thanks for posting here. As a first suggestion, you can pass any OpenDSS command to the `run_command` function in OpenDSSDirect.py. 

```python
import opendssdirect as dss
dss.utils..run_command("""
Redirect ./path/to/circuit.dss
Solve
New Load.Load123 kW=30
Solve
"""
```

So if there's an example of doing what you want in OpenDSS, you can try it with OpenDSSDirect.py. 

I'll add more examples as notebooks.

sibyjackgrove, 2018-10-09:
> Thanks for posting here. As a first suggestion, you can pass any OpenDSS command to the `run_command` function in OpenDSSDirect.py.
>
> ```python
> import opendssdirect as dss
> dss.utils.run_command(""" [...]

Thanks, I'll try these commands out.