dss-extensions/OpenDSSDirect.py #38: Strange behavior when using Like

NicolasGensollen, 2018-10-16:
### Versions

- [ ] Python version: 3.6
- [ ] Python architecture: x64
- [ ] Operating system and version: OSX
- [ ] OpenDSSDirect.py version number: 0.3.4

@kdheepak, I was writing some tests for DiTTo and I have a weird behavior when creating lines using `Like` and getting their values through OpenDSSdirect. I'm not used to work with `Like` but I don't see the logic in the results I am getting.

Please have a look at this simple example notebook which should clarify what I mean here: [link](https://nbviewer.jupyter.org/gist/NicolasGensollen/44b77264122d9619454d6df2ba3ca1d3/Line%20Like%20not%20working%20properly.ipynb)

PMeira, 2018-10-16:
@NicolasGensollen @kdheepak That seems buggy and it also happens in the official OpenDSS. IIRC, the "Like" property for another component was fixed some weeks ago, but I just tested with the latest official SVN binary and it's still broken for lines.

Personally I'd avoid just using `like` , but if you think this should be fixed I can investigate it -- just note that it would result in different behavior when compared to the official OpenDSS binary (at least until it's fixed there too).

NicolasGensollen, 2018-10-17:
Thanks for having a look @PMeira ! :+1: 
I don't really need to use it (actually I never used it before writing these tests...) and I rarely see opendss models relying on the use of Like. 
Might be better to wait for a fix in OpenDSS before fixing it in here. What do you think @kdheepak ?

PMeira, 2021-01-25:
So, looks like EPRI had deprecated using `like` but it was never documented:

https://sourceforge.net/p/electricdss/discussion/861977/thread/8b59d21eb6/?limit=25#b57c/f668

That explains why the many bugs related to it were never fixed.

closed 2022-03-14 by PMeira