dss-extensions/OpenDSSDirect.py #40: adding LoadShape crashes Python

jd-lara, 2018-11-16:
If I run the line to add an element `dss.LoadShape.New("load")` with a system that runs correctly, it is causing the kernel to crash. I am using Python 3.7.1 and OpenDSSDirect 0.3.4

```Python
An error ocurred while starting the kernel
An unhandled exception occurred at $000000011C0C332B:
EParserProblem: Integer number conversion error for string: "follow"
$000000011C0C332B
```

kdheepak, 2018-11-17:
Ouch, that doesn't look good. Can you share the operating system as well? On my MacOS, I'm unable to replicate the same.

```
Python 3.6.4 |Anaconda, Inc.| (default, Jan 16 2018, 12:04:33)
Type 'copyright', 'credits' or 'license' for more information
IPython 6.5.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import opendssdirect as dss
dss.run
In [2]: dss.LoadShape.New("load")
Out[2]: 2

In [3]: dss.__version__
Out[3]: '0.3.4'

In [4]:
```

kdheepak, 2018-11-17:
Can you try it in a fresh terminal? Similar to what I'm done above?

jd-lara, 2018-11-17:
```Python
Python 3.7.1 (default, Nov  6 2018, 18:46:03)
Type "copyright", "credits" or "license" for more information.

IPython 6.5.0 -- An enhanced Interactive Python.

Restarting kernel... 

 

import opendssdirect as dss

import os

dss.run_command('compile {master}'.format(master='master.dss'))
dss.run_command('New Storage.Battery phases=3 Bus1=240_head kV=24 kw=0 kWrated=750.0 kWhrated=2100.0 %EffCharge= dispmode=follow hourly=batteryshape')
dss.utils.class_to_dataframe('Storage')
Out[1]: 
                %Charge %Discharge %EffCharge %EffDischarge %IdlingkW %Idlingkvar %R  %X %reserve  ...   kWrated  kv kvar like model pf phases spectrum yearly
Storage.battery     100          0         90            90         1           0  0  50       20  ...       750  24    0          1  1      3                

[1 rows x 44 columns]

```

as soon as I run the line `dss.LoadShape.New('load')` the it crashes. 

[image]

kdheepak, 2018-11-17:
Can you search for `follow` in the folder with the .dss files? You can use the following in any ternimal window.

```
grep -ri follow folder_name_here
```

kdheepak, 2018-11-17:
This command is incorrect:

```
dss.run_command('New Storage.Battery phases=3 Bus1=240_head kV=24 kw=0 kWrated=750.0 kWhrated=2100.0 %EffCharge= dispmode=follow hourly=batteryshape')
```

It returns the following error:

```
Warning: Duplicate new element definition: "Storage.Battery"
Element being redefined.
Error 303 Reported From OpenDSS Intrinsic Function:
ProcessCommand
Exception Raised While Processing DSS Command:
New Storage.Battery phases=3 Bus1=240_head kV=24 kw=0 kWrated=750.0 kWhrated=2100.0 %EffCharge= dispmode=follow hourly=batteryshape

Error Description:
Floating point number conversion error for string: "dispmode"

Probable Cause:
Error in command string or circuit data.
```

jd-lara, 2018-11-17:
it looks like the storage string is incorrect, but when I run the command to check if a Storage was added it returns the proper data frame. if I run the lines without adding the storage then `dss.LoadShape.New("load")` doesn't cause a crash.

kdheepak, 2018-11-17:
I think the use of `follow` in the storage string is incorrect. Try
removing that keyword and argument and see what you get.

jd-lara, 2018-11-17:
@kdheepak it works ok without adding new things to the simulation. I won't use this command any more since I can't assign to an already loaded file LoadShapes using OpenDSS. I will have to modify the result from the Ditto conversion manually and just use OpenDSSDirect to execute the .dss files.

closed 2018-11-17 by jd-lara