dss-extensions/OpenDSSDirect.py #43: Unable to import opendssdirect

mengyao0412, 2018-11-19:
The version I used is python 2.7 of x64. Following your instruction, the opendssdirect has been successfully installed with 'pip install OpenDSSDirect.py'. However, when I import the library, it encounters an error:
  File "C:\ANACONDA\lib\site-packages\opendssdirect\_utils.py", line 8, in <module>
    from dss.v7.dss_capi import CheckForError

ImportError: No module named dss_capi

Any help will be appreciated.

kdheepak, 2018-11-19:
Was there any error during installation? If there was, can you paste the output of `pip install OpenDSSDirect.py` here?

PMeira, 2018-11-19:
@mengyao0412 Sorry about that. For the moment, with OpenDSSDirect.py v0.3.4, you need to force dss_python v0.9.8 with either: 

`pip install dss_python==0.9.8`

or 

`conda install dss_python==0.9.8 -c pmeira`

This happened because I released a slightly incompatible version of dss_python a couple of days ago. OpenDSSDirect.py v0.3.5 or later will be fine with the new version. Since @kdheepak just merged the changes, so you could just install from git repository if you prefer to update right now.

kdheepak, 2018-11-19:
I figured this might be the issue, so I went ahead with the merge and I added a new version to PyPI just now as well. 

@mengyao0412, try to reinstall using `pip install OpenDSSDirect.py` and everything *should* work.

mengyao0412, 2018-11-19:
@kdheepak @PMeira , I reinstall OpenDSSDirect.py v0.3.5, and everything works. Thank you both for your help!

closed 2018-11-19 by kdheepak