dss-extensions/OpenDSSDirect.py #45: Capacitor and Regulator controls not working

akshaykj92, 2018-12-07:
### Versions

- [ ] Python version: 3.6
- [ ] Operating system and version: Windows 10 and Linux on a High Performance Computing System
- [ ] OpenDSSDirect.py version number: 0.3.5, 0.3.4, 0.3.3

### Bug

* **What is the current behavior?**
We are trying to run some quasi-static time series simulations using opendssdirect.py on a HPC. The controlmode used is "TIME" and we get the same results whether we use the "Duty" or "Yearly" modes. We wish to track the number of regulator and capacitor state changes, which we read directly from the event log. 
When the simulations are run on a Windows PC we see the controller operations both with OpenDSSDirect.py 0.3.3 and 0.3.5 whereas we do not see these controller operations on HPC with both OpenDSSDirect.py versions. Strangely enough we did see these controller operations even on HPC back around October. 
Now there are two possibilities, either the event log is not getting written or the controllers are not operating because of which the event log is empty on HPC. On windows the event log gets written. However I am almost certain that the controllers are not operating. As shown in the graph below of the max and min voltages observed during the entire year on HPC, we see that the voltage never goes above 1.04 which is the substation voltage. However the voltage goes below 0.95 during the peak load summer months which was not the case in our initial runs back in October. We never saw any undervoltages as the cap banks rectified them. 
[image]

Secondly as suggested by @kdheepak I checked the opendss versions on my own system and on HPC and there is a difference. I am not sure what version of OpenDSS was used on HPC back in October though.
Windows system:
0.3.5
DSS C-API Library version 0.10.0 revision 4f258c0c5f23c798d27390e47a9a9fc7e8f8a86f based on OpenDSS SVN 2395 (v7/classic variation) (32-bit build); License Status: Open

HPC
[image]

Finally, when I tried installing opendssdirect.py version 0.3.4 again, and use the command import opendssdirect as dss, I get the following error both on windows and HPC:
[image]

(Sorry for being too verbose, please let me know if any of this doesn't make sense)

PMeira, 2018-12-07:
@akshaykj92 Interesting, we use both types of control at Unicamp without issue. We also do use the event log to extract some information on both Windows and Linux.

> Now there are two possibilities, either the event log is not getting written or the controllers are not operating because of which the event log is empty on HPC.

Good info, it might be some permission issue. 

> Finally, when I tried installing opendssdirect.py version 0.3.4 again, and use the command import opendssdirect as dss, I get the following error both on windows and HPC

You need to downgrade the dependencies, e.g. `pip install dss_python==0.9.8` for older versions.

> is there a way to download x64 versions on dss_python on Windows? That seems to be the only difference at the moment. I can build from source next week.

@kdheepak Sure, the package is available in both my conda channel (pmeira) and PyPI:

https://anaconda.org/PMeira/dss_python/files
https://pypi.org/project/dss_python/#files

I actually recommend using the x64 version, the x86 version is available mostly for backwards compatibility and for people who just can't use x64. I don't upload x86 packages for Linux, there was no demand so far.

I guess we need to create a sample to reproduce that. I'll also be able to work on it next week (finishing some other tasks by Monday). Another point I want to work on is the cross-platform validation with the official COM interface, it's almost ready.

akshaykj92, 2018-12-07:
@PMeira Thanks! I downgraded dss_python to 0.9.8 and opendssdirect version 0.3.4 works perfectly now. As expected it didn't help with the controller operations on HPC. But its a start, please let me know if you need any additional info from my side.

akshaykj92, 2018-12-12:
@kdheepak @PMeira Thanks for all your efforts.  I had been running multiple tests on HPC. So I started by setting up the IEEE 8500 node circuit on HPC and ran a simplified QSTS code on it and the event log was written. I then replaced the files with the feeder I had been using and even then the eventlog was written. 

This made us realize that there was some issue with the way we were specifying the export path directory for the eventlog on the HPC. The path apparently works on our windows system but not on the HPC even though the folder structure is exactly the same. Also OpenDSS is case sensitive on HPC, for instance if I want to redirect a file named "**w**irecodes.dss" then the command "Redirect **W**irecodes.dss" works on windows but not on HPC.  Anyways I don't think there is any issue with opendssdirect.py or opendss. I apologize for any inconveniences.

closed 2018-12-12 by akshaykj92

PMeira, 2018-12-12:
>This made us realize that there was some issue with the way we were specifying the export path directory for the eventlog on the HPC. The path apparently works on our windows system but not on the HPC even though the folder structure is exactly the same.

@akshaykj92 We had some issues before that were handled in OpenDSSDirect.py v0.3.5 (with DSS Python 0.10.0), I believe they were https://github.com/PMeira/electricdss-src/issues/3 and https://github.com/NREL/OpenDSSDirect.py/issues/31

I think the best way to handle file paths in the .DSS scripts for multi-platform usage would be to use relative paths, but that's not ideal.

> Also OpenDSS is case sensitive on HPC, for instance if I want to redirect a file named "wirecodes.dss" then the command "Redirect Wirecodes.dss" works on windows but not on HPC.

Yes, that actually depends on the file system, some can be mounted as case insensitive on Linux (usually not recommended). Unfortunately I noticed there are some situations (e.g. in loadshapes) that OpenDSS ignores non-existent files and doesn't seem to log any error. We should add some extra validation soon (tracking it in https://github.com/PMeira/dss_capi/issues/15).

> I will close the issue for now and maybe open it again if there are any updates.

Sure, feel free to reopen if something strange pops up, and thanks for reporting back!

akshaykj92, 2018-12-13:
Thanks @PMeira  I will try these suggestions, hopefully things will work.

akshaykj92, 2018-12-14:
@kdheepak @PMeira : So the issue is finally resolved. Basically what was happening is that if we specify the export path for the eventlog on HPC, then the entire export directory has to be in lowercase. If any of the subfolders in the specified directory has an upper case the event log is not exported and no error shows up either. 

For example: dss.run_command("Export Eventlog Feeder1/Results/Eventlog.csv") doesn't work but dss.run_command("Export Eventlog feeder1/results/eventlog.csv") works. Also its not just that we need to specify everything in lowercase, the directories need to be in lower case as well. If the specified path is in lower case but the actual directory is in upper case then it creates an error as the directory is not found. 

Thanks for all the help!!