dss-extensions/OpenDSSDirect.py #48: Documentation request: Different API, Need mapping from Windows version

annonch, 2019-01-29:
### Documentation Request

In OpenDSS, I use python over the com port. I was able to import the OpenDSSDirect module into my python program but the API is completely different. I was anticipating that this is a Linux compatible plug-and-play replacement for OpenDSS (windows). I am fine with changing the attribute names but I cant even find the setActiveCircuit command in this module. Can someone please create a document that can describe the differences, or point me to one. Additionally the current API doc is too brief https://nrel.github.io/OpenDSSDirect.py/opendssdirect.html

kdheepak, 2019-01-29:
Hi @annonch good question. The current API documentation is definitely sparse, and one of the things I've been meaning to do is add more detail to the documentation. 

Thanks to all the great work done by @PMeira, you can use the COM interface on Mac and Linux as well. I haven't personally tested or used this but I think it's a fairly accurate replication.

```
import opendssdirect as dss
dss._utils.dss_py.use_com_compat(True)
DSS = dss._utils.dss_py.DSS
```

See example screenshot here.

[image]

This functionality is provided by the [`dss_python`](https://github.com/PMeira/dss_python) package used in this package.

Alternatively, you can use the direct interface. The [`tests`](https://github.com/NREL/OpenDSSDirect.py/blob/012954b48a47c6f9120f62e2165fd40711592869/tests/test_opendssdirect.py) are fairly complete and give you a good idea of how to use a function and what to expect when you use one. 

If you have any additional questions feel free to post them here.

annonch, 2019-01-29:
Ohh I see. I am going to try out the com interface and report back. If I am successful I can help with some small tutorial.

PMeira, 2019-01-31:
@annonch If you find any missing method from the COM interface, let me know and I'll take care of implementing it!

annonch, 2019-01-31:
Thanks, it works perfect.

kdheepak, 2019-02-12:
I'm closing this one. Feel free to reopen this issue or open a new one if you have any other questions.  

And if you wish to contribute a tutorial to improve documentation that would be greatly appreciated!

closed 2019-02-12 by kdheepak

PMeira, 2022-12-03:
Since some people get here from e.g. Google results, let me drop this link here which provides most of the mapping and some more info:
 
https://github.com/dss-extensions/dss-extensions/blob/main/python_apis.md