dss-extensions/OpenDSSDirect.py #49: IEEE13 Bus power flow

jeverdave, 2019-02-09:
### Versions

- [ ] Python version:<3.7>
- [ ] Python architecture: 
- [ ] Operating system and version: 

### Feature Request

### Bug

* **What is the current behavior?**

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.**

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

### Support Question

jeverdave, 2019-02-09:
hi guyz.. i am a beginner in programming. I am doing my thesis right now. It is about short circuit fault analysis of IEEE13 Bus using python. Prior to the short circuit, i need to simulate a power flow of 13 Bus using python. Does anyone know how to use the opendssdirect.py in simulating or solving power flow. thank you for the positive response.

jeverdave, 2019-02-09:
or does anyone know where can I get a tutorial of opendssdirect.py?

kdheepak, 2019-02-09:
Hi @jeverdave, good question. Have you used OpenDSS through the GUI on Windows? How familiar are you with using OpenDSS? Additionally, you say you are a beginner in programming, how much have you used Python? Having prior knowledge of OpenDSS and prior knowledge of Python will greatly help in using this package. There's some helpful resources for OpenDSS over [here](https://sourceforge.net/p/electricdss/code/HEAD/tree/trunk/Doc/) and lots of good Python resources over [here](https://docs.python-guide.org/intro/learning/).

The short answer is that if there's anything you can do using the text interface in the OpenDSS GUI on Windows, you can pretty much type in the exact same thing in OpenDSSDirect.py using the `dss.run_command`) function and it'll do the same thing. There's an example in the notebooks folder over [here](https://github.com/NREL/OpenDSSDirect.py/blob/012954b48a47c6f9120f62e2165fd40711592869/docs/notebooks/GettingStarted.ipynb), and there's some more notebooks in that folder. Have you taken a look at that? They do assume you have some OpenDSS and Python background.

Let me know if you need more information. I'm happy to add additional documentation, I've been meaning to for a while and just need to get around to it.

kdheepak, 2019-02-09:
For completeness sake, here's a 3 line example that solves the 13 node circuit:

```python
import opendssdirect as dss
dss.run_command("Redirect /path/to/OpenDSSDirect.py/tests/data/13Bus/IEEE13Nodeckt.dss");
dss.run_command("Solve")
```

jeverdave, 2019-02-09:
Thank you for this advice. It truly help me to understand the program.

kdheepak, 2019-02-12:
Closing. [See forum](https://sourceforge.net/p/electricdss/discussion/) for more detailed information on how to use OpenDSS.

closed 2019-02-12 by kdheepak

Ethisham, 2020-06-17:
@kdheepak  Hi!
I hope you are doing well.
I am preparing for project in which I have to control OpenDSS through python and want to connect an algorithm to opendss, can you please share some links where I can learn from start how to do it?

PMeira, 2020-06-23:
@Ethisham For OpenDSSDirect.py, you just install the module, then import it on your code. You don't need to install OpenDSS since we package our own implementation. You interact with the OpenDSS engine either via DSS text commands, or through the various specific modules.

See https://dss-extensions.org/ for context of our projects. Check this module's docs at https://dss-extensions.org/OpenDSSDirect.py/ and the main differences in our implementation are currently documented at https://github.com/dss-extensions/dss_capi/blob/0.10.x/docs/known_differences.md

There are generaal examples in the official [OpenDSS forum](https://sourceforge.net/p/electricdss/discussion/) and [its documentation](https://sourceforge.net/p/electricdss/code/HEAD/tree/trunk/Version8/Distrib/Doc/).  The official code repository, partially cloned at https://github.com/dss-extensions/electricdss-tst/tree/master/Distrib/Examples, also contains both general and Python examples. The examples there use the official COM API, but we provide a drop-in replacement for that via [DSS Python](https://github.com/dss-extensions/dss_python/) -- you can use it together with ODD.py.

As a rule of thumb, since we're not affiliated with EPRI, avoid questions specific to OpenDSSDirect.py (or other projects from DSS Extensions) on the official OpenDSS forum. I'm pretty sure you'd be welcome to discuss general aspects of OpenDSS there though. 

A quick tip: the official forum has extensive discussions on various topics. Unfortunately, the SourceForge search is very limited. I recommend using a general search engine like Google or DuckDuckGo to search it, e.g. https://www.google.com/search?hl=en&q=site%3Ahttps%3A%2F%2Fsourceforge.net%2Fp%2Felectricdss%2F%20python

(By the way, in general, try to avoid commenting on closed issues on GitHub -- by default they don't show up on the list)