dss-extensions/OpenDSSDirect.py #51: Documentation request: short circuit fault analysis of IEEE13 Bus

jeverdave, 2019-02-11:
### Versions

- [ ] Python version: <3.7->
- [ ] Python architecture: 
- [ ] windows10->
- [ ] OpenDSSDirect.py version number: <!-

### Feature Request

### Bug

* **What is the current behavior?**

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.**

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

### Support Question

jeverdave, 2019-02-11:
I want to determine the short circuit current due to the bus fault of ieee13 node. How do I determine this using opendssdirect?

kdheepak, 2019-02-12:
Have you done a short circuit analysis using the IEEE13 node feeder in OpenDSS? Any command that you can run in OpenDSS, you can run the same command using OpenDSSDirect.py using the `run_command` function. 

```
Python 3.6.7 |Anaconda, Inc.| (default, Oct 23 2018, 14:01:38)
Type 'copyright', 'credits' or 'license' for more information
IPython 6.2.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import opendssdirect as dss

In [2]: dss.run_command("Redirect ./tests/data/13Bus/IEEE13Nodeckt.dss")
Out[2]: ''

In [3]: dss.run_command("new fault.testflt bus1=650 phases=3")
Out[3]: ''

In [4]: dss.run_command("set mode=faultStudy  ! Use this line for fault study")
Out[4]: ''

In [5]: dss.run_command("Solve")
Out[5]: ''

In [6]: dss.run_command("Export faultStudy")
Out[6]: '/Users/$USER/Documents/GitRepos/OpenDSSDirect.py/IEEE13Nodeckt_EXP_FAULTS.CSV'
```

If you have questions about how to use OpenDSS, there's lots of useful information in the forum threads over [here](https://sourceforge.net/p/electricdss/discussion/).

jeverdave, 2019-02-12:
Thank you for your response. I just want to know if is it possible to
conduct a short circuit fault analysis of IEEE13 node in symmetrical
component domain? Thank you

tshort, 2019-02-12:
To me, it sounds like you’re asking others to help you with homework. Please read the main OpenDSS docs better, and try things out in OpenDSS. This repository is for the Python interface not for general OpenDSS questions. Ask on the OpenDSS forum if you need help on that. 

Suggest closing this issue and the other like this one.

kdheepak, 2019-02-12:
Closing. [See forum](https://sourceforge.net/p/electricdss/discussion/) and ask questions there for more detailed information regarding this.

closed 2019-02-12 by kdheepak