dss-extensions/OpenDSSDirect.py #58: No Push method in Control queue?

vermouth1992, 2019-02-23:
I am new to OpenDSS. I look at the examples http://svn.code.sf.net/p/electricdss/code/trunk/Distrib/Doc/Python-to-OpenDSS Control Interface.pdf and there is a Push function of the Control Queue. But it is not implemented in dss_python and this repo. I just wonder how to add action to control queue?

PMeira, 2019-02-24:
Hi @vermouth1992, 
Thanks for reporting, indeed the `Push` function is missing, it was absent from the Pascal library. 

I added it now, we should release an updated package soon (tomorrow or so) with a couple more updates after everything is properly tested.

PMeira, 2019-03-01:
@vermouth1992 A little delayed but DSS Python 0.10.2 was just uploaded to PyPI and contains the missing `Push` method. I ported a VBA sample code [here](https://github.com/dss-extensions/electricdss-tst/blob/master/Test/CtrlQueueTest.py).

We can leave this issue open until we add the equivalent method in OpenDSSDirect.py.

vermouth1992, 2019-03-01:
Thank you for the update! By the way, I have a general concern about OpenDSS's exception system. It seems that if I call SetActiveBus or SetActiveElement on some non-exist bus or element, the program will not throw an exception but continue running. This behavior makes debugging really difficult.

kdheepak, 2019-03-01:
@vermouth1992 I had the same concern. I think we can change this behavior such that if it returns `0` it raises an exception.

PMeira, 2019-03-01:
@vermouth1992 Yeah, it's a bit messy. We're slowly writing more error handling code to include more exceptions. For `SetActiveBus`/`SetActiveBusi`/`SetActiveElement`, it returns `-1` on invalid names/numbers. Other functions return `0`, an empty string, sometimes `[0]`. Part of the OpenDSS code use the `Error` interface to pass an error code, which are partially mapped to exceptions in DSS Python. 

For DSS Python we cannot really change the default behavior right now (would break COM compatibility), but a flag to use exceptions on these cases is useful. For OpenDSSDirect.py, exceptions seem more natural for all errors.

> I had the same concern. I think we can change this behavior such that if it returns `0` it raises an exception.

@kdheepak Agreed.

closed 2019-03-28 by vermouth1992