dss-extensions/OpenDSSDirect.py #59: Documentation request: IEEE13 Bus Y matrix in sequence or symmetrical component frame

jeverdave, 2019-02-23:
### Versions

- [ ] Python version: 3.7
- [ ] Python architecture:  x64
- [ ] Operating system and version: Windows 10
- [ ] OpenDSSDirect.py version number: v0.3.5

### Feature Request

### Bug

* **What is the current behavior?**

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.**

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

### Support Question

jeverdave, 2019-02-23:
i just want some help on what is the code on how to get the Y matrix of IEEE13 node in sequence or symmetrical frame.

kdheepak, 2019-02-23:
```
from scipy.sparse import csc_matrix
import opendssdirect as dss

...

Y = csc_matrix(dss.YMatrix.getYsparse())
```

I'm on my mobile, and I'm typing this from memory. Excuse any typos or mistakes.

jeverdave, 2019-02-23:
thank you for the code but when i put the code in the interpreter nothings
show up like the Y bus of IEEE 13 bus. how to show or display the Y matrix?
[image: pics.png]

kdheepak, 2019-02-23:
Did you load the 13 bus system before getting the YMatrix?

```
dss.run_command("compile path/to/13bus.dss")
```

jeverdave, 2019-02-23:
yes I loaded the 13 bus system.
dss.run_command('Redirect ../../tests/data/13Bus/IEEE13Nodeckt.dss');
but nothings happen

kdheepak, 2019-02-23:
What happens when you `print(csc_matrix(dss.YMatrix.getYsparse())` after you load the system?

jeverdave, 2019-02-23:
Thank you it shows up already..may I know if it is per sequence like
positive, negative, and zero sequennce admittance matrix

PMeira, 2019-02-23:
@jeverdave By default OpenDSS uses the full phase model. If you want a positive sequence equivalent, you could do:
```
dss.run_command('MakePosSeq') 
dss.run_command('Solve') # this is needed so the matrix is rebuilt

print(csc_matrix(dss.YMatrix.getYsparse())
```

There are caveats, check [the manual](https://sourceforge.net/projects/electricdss/files/OpenDSS/OpenDSSManual.pdf/download) for some more info, and of course there is the official OpenDSS forums with past discussions: https://sourceforge.net/p/electricdss/discussion/

For negative and zero sequences, you'd need to process the full matrix yourself. OpenDSS doesn't really use them for the solution, all input parameters are converted to phase components. Depending on the system, this is not trivial.

jeverdave, 2019-02-24:
Thank you sir. 😁 I will surely apply this.

closed 2019-02-24 by PMeira