dss-extensions/OpenDSSDirect.py #61: Documentation request: zero sequence admittance matrix of IEEE13

jeverdave, 2019-03-01:
### Versions

- [ ] Python version: <3.7>
- [ ] Python architecture: < x64>
- [ ] Operating system and version: <windows 10>
- [ ] OpenDSSDirect.py version  

### Feature Request

### Bug

* **What is the current behavior?**

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.**

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

### Support Question

jeverdave, 2019-03-01:
I am trying to form zero sequence admittance matrix o IEEE13 Bus but I dont know how using opendssdirect. I need some help. I am really a beginner in programming ang I am new with the software.

tshort, 2019-03-01:
This is not the OpenDSS help line. Issues are meant for bug reports, feature requests, and other development-related discussions. Please don't use it for help-line type questions.

jeverdave, 2019-03-02:
Im very sorry for always asking for help but can you help me just this one?
Please.. just a small guide..

PMeira, 2019-03-03:
@jeverdave I'll quote what I said in #59 

> By default OpenDSS uses the full phase model. If you want a positive sequence equivalent, you could do:
> ```python
> dss.run_command('MakePosSeq') 
> dss.run_command('Solve') # this is needed so the matrix is rebuilt
> print(csc_matrix(dss.YMatrix.getYsparse())
> ``` 
> There are caveats, check the manual for some more info, and of course there is the official OpenDSS forums with past discussions: https://sourceforge.net/p/electricdss/discussion/
> 
> For negative and zero sequences, you'd need to process the full matrix yourself. OpenDSS doesn't really use them for the solution, all input parameters are converted to phase components. Depending on the system, this is not trivial.
> 

Trying to make it clear: no, there is no **zero sequence admittance** for the full system in OpenDSS. However, you COULD compute it from the `dss.YMatrix.getYsparse()`. The issue is, OpenDSS is a multi-phase solver, you can have any number or configuration of phases.

My suggestion is: 
1) grab a book and learn you to do the transformation yourself
2) get `csc_matrix(dss.YMatrix.getYsparse()`
3) process `dss.Circuit.AllNodeNames()` to assign phases (this might be system specific)
4) what now? Well, go study and figure it out yourself.

I mentioned this...

> There are caveats, check the manual for some more info, and of course there is the official OpenDSS forums with past discussions: https://sourceforge.net/p/electricdss/discussion/

...because other people have asked similar questions there. I don't think there is a full solution but you might learn something.

**So, again, please stop posting this kind of question here. Any further posts like this will likely be closed without any comments.**

I'll close and lock this now.

closed 2019-03-03 by PMeira