dss-extensions/OpenDSSDirect.py #62: Segfault when using `dss.utils.class_to_dataframe("transformer")`

kdheepak, 2019-03-01:
### Versions

- Python version: 3.6
- Python architecture: x64
- Operating system and version: MacOS
- OpenDSSDirect.py version number: v0.3.6

### Bug

* **What is the current behavior?**

It segfaults when running `class_to_dataframe("transformer")`

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**

Ideally no function would ever segfault

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.**

```
$ cat master.dss

clear

new circuit.IEEE13Nodeckt
~ basekv=115 pu=1.0001 phases=3 bus1=SourceBus
~ MVAsc3=20000 MVASC1=21000    ! stiffen the source to approximate inf source

New Transformer.Sub Phases=3 Windings=2

$ ipython
Python 3.6.5 |Anaconda, Inc.| (default, Mar 29 2018, 13:14:23)
Type 'copyright', 'credits' or 'license' for more information
IPython 6.5.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import opendssdirect as dss

In [2]: dss.run_command("Redirect master.dss")
Out[2]: ''

In [3]: dss.Circuit.SetActiveClass("transformer")
Out[3]: 21

In [4]: dss.ActiveClass.Name("transformer.sub")

In [5]: dss.ActiveClass.Name()
Out[5]: 'sub'

In [6]: len(dss.Element.AllPropertyNames())
Out[6]: 55

In [7]: dss.Element.AllPropertyNames()[45]
Out[7]: 'Core'

In [8]: dss.Element.AllPropertyNames()[44]
Out[8]: 'WdgCurrents'

In [9]: dss.Properties.Value(str(45 + 1))
Out[9]: 'shell'

In [11]: dss.Properties.Value(str(44 + 1))
[1]    96353 segmentation fault  ipython
```

kdheepak, 2019-03-01:
cc @tarekelgindy

PMeira, 2019-03-01:
On Windows that didn't segfault, but it seems invalid properties are indeed crashing.

By the way, I don't think we need the `str` here, it gets converted back to integer:

> dss.Properties.Value(str(45 + 1))

You can either by index (`dss.Properties.Value(45 + 1)`) or name (`dss.Properties.Value('Core')`)

PMeira, 2019-03-01:
With the sample code, I got the issue on Windows x86 with `dss_python==0.10.0`. Upgrading to `dss_python==0.10.1` or `dss_python==0.10.2`  works. @kdheepak Can you check if that works on macOS?

(We still need to fix the crash with invalid properties.)

kdheepak, 2019-03-01:
Upgrading to `dss_python==0.10.1` resolves this!

kdheepak, 2019-03-01:
I guess if I tag a release on the latest master and update PyPI this can be closed.

PMeira, 2019-03-01:
Nice!
I left a more detailed analysis of the remaining issue (invalid properties) in https://github.com/dss-extensions/dss_capi/issues/38 -- we can work around it in Python if needed by checking if the current property name is not empty before getting the value.

kdheepak, 2019-03-04:
This should be resolved in [v0.3.7](https://github.com/dss-extensions/OpenDSSDirect.py/releases/tag/v0.3.7).

closed 2019-03-04 by kdheepak