dss-extensions/OpenDSSDirect.py #63: dss.Generators kW function is missing

julioolvera1, 2019-03-05:
dss.__version__
Out[497]: '0.3.7'

When using the following loop to get the functions for the generators, the "kW" funciton is missing.

for name, function in inspect.getmembers(dss.Generators):
    if callable(function) and not name.startswith('_'):
        print(f'dss.Generators.{name}')
        
        DSSgens_fcns.append(name)
[image]

PMeira, 2019-03-06:
Since you didn't provide the info requested in issue template, there's not much we can do.

```
- Python version: <!-- (e.g. 2.7, 3.3, 3.6) -->
- Python architecture: <!-- (e.g. x86, x64) -->
- Operating system and version: <!-- (e.g. Windows XP, Windows 7, Windows 8, OSX 11, Red Hat, Ubuntu, Centos 6, Centos 7) -->
```

It works for me, a complete sample:

```python
import inspect
from pprint import pprint
import opendssdirect as dss

for name, function in inspect.getmembers(dss.Generators):
    if callable(function) and not name.startswith('_'):
        print(f'dss.Generators.{name}')

print()
print(dss.__version__)
```

Output:
```
dss.Generators.AllNames
dss.Generators.Count
dss.Generators.First
dss.Generators.ForcedON
dss.Generators.Idx
dss.Generators.Model
dss.Generators.Name
dss.Generators.Next
dss.Generators.PF
dss.Generators.Phases
dss.Generators.RegisterNames
dss.Generators.RegisterValues
dss.Generators.Vmaxpu
dss.Generators.Vminpu
dss.Generators.get_float64_array
dss.Generators.get_string
dss.Generators.get_string_array
dss.Generators.kV
dss.Generators.kVARated
dss.Generators.kW
dss.Generators.kvar

0.3.7
```

Don't know why you'd do that though, there is `dss.Generators.__all__` already.

julioolvera1, 2019-03-06:
Thank you for your reply. The `dss.Generators.__all__` does have the kW property.
However, I have another problem, when I try to assign/modify the original kW of a previously defined generator, it does not get modified. I followed the same process as I would do using the COM.

`import opendssdirect as dss`

`DSSGens=dss.Generators`
`MyGenNames=DSSGens.AllNames()`

`DSSGens.First()`
`for ii in range(0,len(MyGenNames):`
        `DSSGens.kW=float(P_VSC_inj_F3[ii]*1e3)`
        `DSSGens.Next()`

PMeira, 2019-03-06:
@julioolvera1 One possibility is that your installation is broken somehow. There have been no recent changes about the generators and the `kW` function exists and is listed in `__all__` -- I checked the PyPI package, and here it is in the repo:

https://github.com/dss-extensions/OpenDSSDirect.py/blob/64e03a111896c928208f8e3d23149632b72f9222/opendssdirect/Generators.py#L205

But there could be another issue:
```
DSSGens.kW=float(P_VSC_inj_F3[ii]*1e3)
```
This is invalid, you'd be overwriting the `kW` function with a float (hence the `TypeError` later). For OpenDSSDirect.py, the correct would be 
```
DSSGens.kW(float(P_VSC_inj_F3[ii]*1e3))
```

If you already have code that uses the COM syntax, you might prefer to use [DSS Python](https://github.com/dss-extensions/dss_python) (you already have it installed, it's a dependency) since it tries to be very compatible with COM. You can still import OpenDSSDirect.py and use it together, including `generators_to_dataframe`.

julioolvera1, 2019-03-07:
I see, I think that is the root of the problem. I was overwriting the kW function instead of assigning a new value. 

I haven't considered the option of using DSS Python. In your opinion, are there any advantages between OpenDSSDirect and DSS Python? Is there a difference in performance or it's just that the syntax is more COM-like?

Thank you for your help.

PMeira, 2019-03-07:
> In your opinion, are there any advantages between OpenDSSDirect and DSS Python? Is there a difference in performance or it's just that the syntax is more COM-like?

It's mainly the syntax. ODD.py has a couple of extras in the Python side (such as the dataframes, docs). DSSpy has a few extensions/newer classes (not present in COM) that weren't ported to ODD.py yet, as well as mapping from the Error interface to Python exceptions more frequently. These differences should be gone after #55 is done. There is a new batch of extensions cooking in the Pascal side of OpenDSS too.

The only other notable difference (that I can think right now) is that, for backwards compatibility, ODD.py uses lists. DSSpy returns NumPy arrays for most things. But there is a plan to change that soon too (#60).

They can be used together because they bind to the same instance of the DSS C-API library. Besides being multi-platform, the library itself removes a lot of the overhead from the COM interface and we're slowly making changes to add more error checking and handle performance issues. So, in the end, both Python modules are generally faster (sometimes several times) than the COM alternative.

closed 2020-03-11 by PMeira