dss-extensions/OpenDSSDirect.py #65: Getting: "TypeError: 'str' object is not callable"

JonasVil, 2019-03-25:
Versions

- [ ] Python version: 3.7
- [ ] Python architecture:  x64
- [ ] Operating system and version: Windows 10 Pro
- [ ] OpenDSSDirect.py version number: v0.3.7

Hello everyone, my problem is the following, when I am trying to run my simulations I keep getting the following message:

TypeError: 'str' object is not callable

For the following line of code:

DSS.Text.Command('Redirect ../OpenDSS_files/Master.dss')

The first time I run the simulation, it works perfectly. However, when I run it again, it identifies the error. What could be happening? Thank you all.

kdheepak, 2019-03-25:
I think you are using the COM interface. Try the following:

```python
DSS.Text.Command = 'Redirect ../OpenDSS_files/Master.dss'
```

kdheepak, 2019-03-25:
> The first time I run the simulation, it works perfectly.

This is odd though? However, without looking at exactly what you are doing I don't know if I'll be able to offer any additional information. When you say "the first time I run the simulation" what do you mean? When you "run it again", you are not restarting the Python interpreter, right? If the first thing you do is initialize the COM interface, you should not have different behavior when you run it multiple times given that you don't close the Python interpreter.

Feel free to post here if the solution in the previous does not work.

JonasVil, 2019-03-27:
Hi,

Thanks for the feedback, I tried to use it the way you indicated, but it did not work. I rewrote the code, using different variables, with the same syntax, and finally, the simulation worked. Thank you so much for your help.

Another problem appeared, my simulation basically makes decisions on how the storage system will operate in the next time interval for one year divided into 1-hour intervals. Although with the use of "OpenDSSDirect.py" the simulation time has decreased substantially, I could notice that the ram memory consumption is too high, the simulation ends with a consumption of 2 Gb of RAM.

Evaluating the code I can conclude that the accumulation of RAM happens when the decision is made and solve the load flow for the respective time interval. I wonder if there is a way to get this memory released while the simulation happens.

kdheepak, 2019-03-27:
You should not be having memory issues if you are using the latest version of OpenDSSDirect.py. 

To identify where the issue is occuring, after running the simulation for multiple hours, open up a task manager and look at the memory consumption of Python. Then, run `opendssdirect.run_command("clear")` in Python and see if the memory consumption changes. Your code should look something like this:

```
import opendssdirect
# run your code here for initialization
for t in 1:8760: # assuming you are running 8760 time periods
    # ... run your code here for every hour
    if t == 500:
        print("Clearing memory")
        opendssdirect.run_command("clear")
```

Does that make sense?

If you see the memory drop down significantly then it means that there's some memory being allocated in OpenDSSDirect.py that is not being deallocated in each loop. If not, then you may be storing large objects in memory in your code and the Python garbage collector is not releasing them.

Hope that makes sense? Without more details (exact code, memory consumption stats, running a memory profiler on your machine etc) it will be hard for us to debug this.

PMeira, 2019-03-27:
> Evaluating the code I can conclude that the accumulation of RAM happens when the decision is made and solve the load flow for the respective time interval.

@JonasVil To complement @kdheepak's suggestion, I imagine you use Monitors, right? Do you clear/reset them periodically? If not, calling `opendssdirect.Monitors.ResetAll()` after grabbing the data could be enough.

Recently, a colleague has been testing dynamic simulations with time steps of 1 ms and, so far, we have not seen memory leaks.

> with the use of "OpenDSSDirect.py" the simulation time has decreased substantially

That's nice!

JonasVil, 2019-03-27:
@kdheepak I had already tried this approach. I use the "External" mode of the OpenDSS storage system. Through this mode it is possible to change in which state the storage will operate in the next step of time, in this way, I change the state in each interaction using the command "edit storage", being necessary to load only once the "master.dss". If I run the "clear" command just as you suggested, I lose all system information loaded in the previous interactions.

@PMeira In the early versions of the code, I used monitors to extract the information that was needed. Because I read some reports about how using monitors could increase simulation time, I looked for ways to obtain information from the system in a way that did not consume so much time. Using "AllVariableValues" was possible to improve the performance of my code.

Thank you both for the help!

JonasVil, 2019-03-27:
Here is an example of when memory leak is seen:

`import opendssdirect as DSS
import numpy as np

DSS.Text.Command('Redirect ../OpenDSS_files/Master.dss')
DSS.Text.Command("New Storage.{storage_name} Bus1={bus_name} phases=3 kV=12.66 kWRated={rating} kWhRated={kwh_rating} %stored={initial_state} %IdlingkW=1 %reserve=35 %EffCharge=95 %EffDischarge=95 State=idling dispmode=External model=1".format(
    storage_name='Storage1',
    bus_name='k2',
    rating=1000,
    kwh_rating=1000,
    initial_state=35
    ))
DSS.Text.Command('set mode=yearly')
DSS.Text.Command('set stepsize=1h')
DSS.Text.Command('set number = 1000000')
DSS.Text.Command('Edit Storage.Storage1 state=idling')
DSS.Text.Command('set number=1')
DSS.Solution.Solve()

SoC = np.zeros((1000001, 1))
DSS.Circuit.SetActiveElement('Storage1')
Storage_State = DSS.CktElement.AllVariableValues()
SoC[0] = Storage_State[0]

for i in range(1000000):
    if SoC[i] < 1000:
        DSS.Text.Command('Edit Storage.Storage1 state=charging %charge=100')
    
        
    elif SoC[i] == 1000:
        DSS.Text.Command('Edit Storage.Storage1 state=discharging %discharge=100')
        
    DSS.Text.Command('set number=1')
    DSS.Solution.Solve()
    DSS.Circuit.SetActiveElement('Storage1')
    Storage_State = DSS.CktElement.AllVariableValues()
    SoC[i+1] = Storage_State[0]`

JonasVil, 2019-03-27:
Here is an example of when memory leak is seen:

```
import opendssdirect as DSS
import numpy as np

DSS.Text.Command('Redirect .../OpenDSS_files/Master.dss')
DSS.Text.Command("New Storage.{storage_name} Bus1={bus_name} phases=3 kV=12.66 kWRated={rating} kWhRated={kwh_rating} %stored={initial_state} %IdlingkW=1 %reserve=35 %EffCharge=95 %EffDischarge=95 State=idling dispmode=External model=1".format(
    storage_name='Storage1',
    bus_name='k2',
    rating=1000,
    kwh_rating=1000,
    initial_state=35
    ))
DSS.Text.Command('set mode=yearly')
DSS.Text.Command('set stepsize=1h')
DSS.Text.Command('set number = 1000000')
DSS.Text.Command('Edit Storage.Storage1 state=idling')
DSS.Text.Command('set number=1')
DSS.Solution.Solve()

SoC = np.zeros((1000001, 1))
DSS.Circuit.SetActiveElement('Storage1')
Storage_State = DSS.CktElement.AllVariableValues()
SoC[0] = Storage_State[0]

for i in range(1000000):
    if SoC[i] < 1000:
        DSS.Text.Command('Edit Storage.Storage1 state=charging %charge=100')
    
        
    elif SoC[i] == 1000:
        DSS.Text.Command('Edit Storage.Storage1 state=discharging %discharge=100')
        
    DSS.Text.Command('set number=1')
    DSS.Solution.Solve()
    DSS.Circuit.SetActiveElement('Storage1')
    Storage_State = DSS.CktElement.AllVariableValues()
    SoC[i+1] = Storage_State[0]
```

kdheepak, 2019-03-27:
Thanks for the example, that was useful to see.

I changed the `for` loop to this:

```python
for i in range(500000):
    DSS.Solution.Solve()
    Storage_State = DSS.CktElement.AllVariableValues()
    SoC[i + 1] = Storage_State[0]
```

and I was still observing an increase in memory (from 43M to 47M). However, if I changed the same for loop to the following:

```python
for i in range(500000):
    DSS.Solution.Solve()
    DSS.CktElement.AllVariableValues()
    # SoC[i + 1] = Storage_State[0]
```

Then it stays at a constant 43M, i.e. no memory leak. This makes me think it's a Python garbage collection and/or NumPy issue. However, when I added the `Edit Storage.Storage1 state=idling` it always increases in memory, even if I don't use the `SoC[i + 1] = Storage_State[0]` 

The above examples took about 15 seconds to run on my machine. I used the following to get the memory of the process. This runs it in the background and monitors the process in the current terminal.

```bash
top -pid `{ python test.py > /dev/null & } && echo $!`
# press q to exit
killall python # run this to kill the background process.
```

I was running this on the 13Bus test case in this repo in the tests/data folder.

kdheepak, 2019-03-27:
I'm pretty sure OpenDSS cache's previous results, so unless something changes OpenDSS is going to avoid doing extra work.

kdheepak, 2019-03-27:
@PMeira any other insight you can provide here would be valuable. I'm not sure what the cause of the memory leak here is.

PMeira, 2019-03-27:
@JonasVil Like Dheepak said, the example really helped, thank you!

@kdheepak Yep, looks like a NumPy issue with element access. Here's a modified code sample:

```python
import numpy as np 

USE_NUMPY = True
DO_COPY = False
TOTAL_STEPS = 10000000
NUM_BLOCKS = 10 
BLOCK_SIZE = (TOTAL_STEPS // NUM_BLOCKS)

def run_block(i_start, SoC, Storage_State):
    for i in range(i_start, i_start + BLOCK_SIZE):
        SoC[i+1] = Storage_State[0]

    if DO_COPY:
        return np.copy(SoC)
        
    return SoC

def run():
    if USE_NUMPY:
        print(np.__version__)
        
        SoC = np.zeros((TOTAL_STEPS + 1, 1))
    else:
        SoC = [0.0] * (TOTAL_STEPS + 1)
    
    Storage_State = [350.00000000000006, 0.0, 0.0, 0.0, 1.1890276558759354, 1.1890276558759354, 0.0]

    for i in range(NUM_BLOCKS):
        print('Block', i)
        SoC = run_block(i * BLOCK_SIZE, SoC, Storage_State)
        input()
        
    print('Done 1') # Memory is still held here, probably in `SoC`
    input()

run()
print('Done 2') # Here the memory is already freed
input()
```

Observations:
- setting `USE_NUMPY = True` and `DO_COPY = False` increases memory usage over time.
- setting `USE_NUMPY = True` with `DO_COPY = True` forces the old data to be garbage collected and memory usage seems stable -- so this is a workaround.
- settings `USE_NUMPY = False` (using a list instead) uses more memory but doesn't grow.

I find this really unexpected.

For NumPy 1.8.2, 1.15.4, and 1.16.2 (latest), the "leaky" behavior was the same.

I tested with NumPy 1.6.2 (the oldest available in conda) and 1.7.1 and it doesn't seem to grow/leak -- if grows momentarily but shrinks back. 

Just this:
```
    for i in range(i_start, i_start + BLOCK_SIZE):
        SoC[i] * 2
```
is actually enough to cause memory growth. The copy still helps in this case.

If this is not reported already in the NumPy repo, we should probably investigate a bit further and report it.

PMeira, 2019-07-26:
@JonasVil @kdheepak Revisiting this now, the apparent memory leak became clear.

Even though there's no mention [in the docs](https://docs.scipy.org/doc/numpy/reference/generated/numpy.zeros.html), `np.zeros` may use lazy allocation, so the memory for the whole array `SoC` is not initially completely allocated. It only gets allocated as we access the elements. It seems to use `calloc` in C which, depending on the OS and memory allocation algorithms, will present this behavior.

To avoid the apparent leak, use `np.full` instead to provide the whole array "preallocated" with a selected value. For long running simulations, I'd recommend it.

closed 2019-07-26 by PMeira