dss-extensions/OpenDSSDirect.py #66: Is there anyway to export the grid structure as a graph (e.g. networkX)?

vermouth1992, 2019-03-28:
### Versions

- [ ] Python version: 
- [ ] Python architecture: 
- [ ] Operating system and version: 
- [ ] OpenDSSDirect.py version number: 

### Feature Request

### Bug

* **What is the current behavior?**

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.**

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

### Support Question

kdheepak, 2019-03-28:
This is a simple script that can get you started.

```python
import opendssdirect as dss
import networkx as nx

dss.run_command("Redirect ./tests/data/13Bus/IEEE13Nodeckt.dss")

G = nx.Graph()

for i, l in enumerate(dss.Lines.AllNames()):
    dss.Lines.Name(l)
    b1 = dss.Lines.Bus1()
    b2 = dss.Lines.Bus2()
    G.add_edge(b1.split(".")[0], b2.split(".")[0])
```

I've written a more generalized version that works across single phase systems including transformers, switches, loads, etc, but I haven't made that public yet.

It might be worth looking at [krangpower](https://github.com/supsi-dacd-isaac/krangpower/) to see if they've implemented what you need.

edxu96, 2020-07-28:
The method by @kdheepak omits transformers at least. I implement a method based on [a discussion about getting elements information using matlab](https://sourceforge.net/p/electricdss/discussion/861976/thread/d47aed64/?limit=25).

```Python
import opendssdirect as dss

# Gather names of all transformers
trans_name = list(dss.utils.transformers_to_dataframe().index)
trans_bus = {}

# Set each transformer as active element and store its two buses.
for t in trans_name:
    dss.Circuit.SetActiveElement('Transformer.' + t)
    trans_bus[t] = dss.CktElement.BusNames()
```

Then you can plot the red circle part for IEEE 1992 13 buses test case, for example.

[image]