dss-extensions/OpenDSSDirect.py #67: broken links when installing

tarekelgindy, 2019-05-15:
I'm having issues installing opendssdirect and can't seem to google my way out of it.
I've created a conda environment `qsts` with anaconda and python 3.6 on a linux HPC machine.

When I ran `pip install opendssdirect.py` I got this error:

```
Exception:
Traceback (most recent call last):
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/download.py", line 809, in unpack_url
    unpack_file_url(link, location, download_dir, hashes=hashes)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/download.py", line 715, in unpack_file_url
    unpack_file(from_path, location, content_type, link)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/utils/__init__.py", line 599, in unpack_file
    flatten=not filename.endswith('.whl')
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/utils/__init__.py", line 484, in unzip_file
    zip = zipfile.ZipFile(zipfp, allowZip64=True)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/zipfile.py", line 1100, in __init__
    self._RealGetContents()
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/zipfile.py", line 1168, in _RealGetContents
    raise BadZipFile("File is not a zip file")
zipfile.BadZipFile: File is not a zip file
```

I tried installing by downloading and installing the tar file but got this error instead:
```
Obtaining file:///home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/OpenDSSDirect.py-0.3.7
Collecting future (from OpenDSSDirect.py==0.3.7)
Requirement already satisfied: six in /home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages (from OpenDSSDirect.py==0.3.7)
Collecting dss_python==0.10.1 (from OpenDSSDirect.py==0.3.7)
  Using cached https://files.pythonhosted.org/packages/17/ed/73d310165eba4a293cf7c346a0676296e0df5dc6a35821f8eed618591156/dss_python-0.10.1-cp36-cp36m-manylinux1_x86_64.whl
Requirement already satisfied: numpy>=1.0 in /home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages (from dss_python==0.10.1->OpenDSSDirect.py==0.3.7)
Collecting cffi>=1.11.2 (from dss_python==0.10.1->OpenDSSDirect.py==0.3.7)
Exception:
Traceback (most recent call last):
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/_vendor/requests/sessions.py", line 488, in get
    return self.request('GET', url, **kwargs)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/_vendor/requests/sessions.py", line 475, in request
    resp = self.send(prep, **send_kwargs)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/_vendor/requests/sessions.py", line 617, in send
    history = [resp for resp in gen] if allow_redirects else []
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/_vendor/requests/sessions.py", line 617, in <listcomp>
    history = [resp for resp in gen] if allow_redirects else []
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/_vendor/requests/sessions.py", line 177, in resolve_redirects
    **adapter_kwargs
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/_vendor/requests/sessions.py", line 596, in send
    r = adapter.send(request, **kwargs)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/_vendor/cachecontrol/adapter.py", line 37, in send
    cached_response = self.controller.cached_request(request)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/_vendor/cachecontrol/controller.py", line 111, in cached_request
    resp = self.serializer.loads(request, cache_data)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/_vendor/cachecontrol/serialize.py", line 114, in loads
    return getattr(self, "_loads_v{0}".format(ver))(request, data)
  File "/home/telgindy/anaconda2/envs/qsts/lib/python3.6/site-packages/pip/_vendor/cachecontrol/serialize.py", line 176, in _loads_v2
    cached = json.loads(zlib.decompress(data).decode("utf8"))
zlib.error: Error -5 while decompressing data: incomplete or truncated stream
```

Any idea what might be causing this? It looks like the file being downloaded is corrupted.
Hopefully I'm just doing something careless here...

kdheepak, 2019-05-15:
Can you paste the results of `which pip` over here?

kdheepak, 2019-05-15:
I was able to do a fresh install just now on my Mac in a new conda environment:

```
$ pip install OpenDSSDirect.py                                                                             
Collecting OpenDSSDirect.py
  Downloading https://files.pythonhosted.org/packages/bd/16/1a45f4deffb2be8d9476f073feab39628de33925131fc753a6a1cb8d4319/OpenDSSDirect.py-0.3.7.tar.gz (71kB)
     |████████████████████████████████| 71kB 1.3MB/s
Collecting future (from OpenDSSDirect.py)
  Downloading https://files.pythonhosted.org/packages/90/52/e20466b85000a181e1e144fd8305caf2cf475e2f9674e797b222f8105f5f/future-0.17.1.tar.gz (829kB)
     |████████████████████████████████| 829kB 2.2MB/s
Collecting six (from OpenDSSDirect.py)
  Using cached https://files.pythonhosted.org/packages/73/fb/00a976f728d0d1fecfe898238ce23f502a721c0ac0ecfedb80e0d88c64e9/six-1.12.0-py2.py3-none-any.whl
Collecting dss_python==0.10.1 (from OpenDSSDirect.py)
  Downloading https://files.pythonhosted.org/packages/47/d2/63142512b3048e2116e71c2f823da7c886de04f2ae4fc47818574d8a6c6e/dss_python-0.10.1-cp36-cp36m-macosx_10_7_x86_64.whl (3.5MB)
     |████████████████████████████████| 3.5MB 3.5MB/s
Collecting numpy>=1.0 (from dss_python==0.10.1->OpenDSSDirect.py)
  Downloading https://files.pythonhosted.org/packages/ae/76/4a4c012bca5688881c18f6e04694d221b88daa5d38526a4df87d75711199/numpy-1.16.3-cp36-cp36m-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl (13.9MB)
     |████████████████████████████████| 13.9MB 3.9MB/s
Collecting cffi>=1.11.2 (from dss_python==0.10.1->OpenDSSDirect.py)
  Downloading https://files.pythonhosted.org/packages/79/46/a0133536162ad7fe5ae27f22f6ed14a77132b32589dce71223ebe7e02371/cffi-1.12.3-cp36-cp36m-macosx_10_6_intel.whl (247kB)
     |████████████████████████████████| 256kB 6.0MB/s
Collecting pycparser (from cffi>=1.11.2->dss_python==0.10.1->OpenDSSDirect.py)
  Downloading https://files.pythonhosted.org/packages/68/9e/49196946aee219aead1290e00d1e7fdeab8567783e83e1b9ab5585e6206a/pycparser-2.19.tar.gz (158kB)
     |████████████████████████████████| 163kB 6.6MB/s
Building wheels for collected packages: OpenDSSDirect.py, future, pycparser
  Building wheel for OpenDSSDirect.py (setup.py) ... done
  Stored in directory: /Users/$USER/Library/Caches/pip/wheels/c0/e7/02/52e95e79425f93b12800cbe2820bf3680c05eae8bc7d561c96
  Building wheel for future (setup.py) ... done
  Stored in directory: /Users/$USER/Library/Caches/pip/wheels/0c/61/d2/d6b7317325828fbb39ee6ad559dbe4664d0896da4721bf379e
  Building wheel for pycparser (setup.py) ... done
  Stored in directory: /Users/$USER/Library/Caches/pip/wheels/f2/9a/90/de94f8556265ddc9d9c8b271b0f63e57b26fb1d67a45564511
Successfully built OpenDSSDirect.py future pycparser
Installing collected packages: future, six, numpy, pycparser, cffi, dss-python, OpenDSSDirect.py
Successfully installed OpenDSSDirect.py-0.3.7 cffi-1.12.3 dss-python-0.10.1 future-0.17.1 numpy-1.16.3 pycparser-2.19 six-1.12.0
```

`OpenDSSDirect.py` depends on `dss-python` which depends on `numpy`. That's the only package that I anticipate causing any wheel building issues (even that seems unlikely on a Windows Linux). Try `pip install dss-python` to see what happens. If you get the same error, try `conda install numpy` and then try `pip install OpenDSSDirect.py` again.

If none of the above work, try `pip install networkx` (which is a pure Python package) to see if there's an issue with pip itself.

tarekelgindy, 2019-05-15:
I was able to debug this. The issue came from a broken install that I'd previously done that left corrupted files in the cache. Specifically there was a file `'/home/telgindy/.cache/pip/wheels/c0/e7/02/52e95e79425f93b12800cbe2820bf3680c05eae8bc7d561c96/OpenDSSDirect.py-0.3.7-cp36-none-any.whl'` which the installer was trying to read which was causing it to crash.

I did `rm -rf .cache/pip/*` and everything worked fine. 
Hope others find this issue useful - this was an annoying one to debug

closed 2019-05-15 by tarekelgindy