dss-extensions/OpenDSSDirect.py #75: Support New OpenDSS Objects?

qlong2, 2020-01-13:
### Versions
- [ ] Python version: 
- [ ] Python architecture: 
- [ ] Operating system and version: 
- [ ] OpenDSSDirect.py version number: 

### Feature Request

Recently the OpenDSS released several new OpenDSS objects for PV system, inverter control, storage and storage controller with new object names (for example, PVSystem2). Some of the issues with original OpenDSS objects are fixed in the new version. However, my python script with the OpenDSSDirect interface seems not to be able to compile the model with these new objects. I wonder if the current version of OpenDSSDirect supports those new objects yet. Thank you.

qlong2, 2020-01-13:
Python version: 2.7.16
Python architecture: x64
Operating system and version: windows 10
OpenDSSDirect.py version number: 0.3.7

PMeira, 2020-01-13:
No, they're no available yet. Adding them right now is not the best idea. 

Those new components were just added (couple of weeks ago, during the holidays). I'd wait a month or two so other users (and EPRI) have time to experiment and provide feedback directly on the official OpenDSS.

PMeira, 2020-01-15:
@qlong2 It seems that even in the upstream OpenDSS those models have not been fully released yet, see https://sourceforge.net/p/electricdss/discussion/experts/thread/20a7da4968/#b9ed

qlong2, 2020-01-18:
@PMeira Your answer makes a lot of sense. Also I found this post very useful. Thank you for your help.

PMeira, 2020-03-02:
@qlong2 The next ODD.py release should include a preview of the new components, but testing them still show some issues even in the latest official OpenDSS -- as such, we can't properly validate them,  consider them experimental. They also don't have dedicate API functions yet.

qlong2, 2020-03-02:
Thank you @PMeira for the update! I understand that the model validation is a time-consuming process. But can't wait to give it a try and thank you for the good work!

> @qlong2 The next ODD.py release should include a preview of the new components, but testing them still show some issues even in the latest official OpenDSS -- as such, we can't properly validate them, consider them experimental. They also don't have dedicate API functions yet.

closed 2020-08-02 by PMeira

PMeira, 2020-08-02:
@qlong2 Just to confirm, the new models are enabled by default since OpenDSS 9.0 and our latest release: https://github.com/dss-extensions/OpenDSSDirect.py/releases/tag/v0.5.0

For our release, you can toggle the old components if necessary, but the default ones are now the new models (previously PvSystem2, Storage2...) just like in OpenDSS 9.

PR #85 (2020-08-02): DSS C-API/Python 0.10.6
- Add CheckForError for most API calls
- Add settings for LegacyModels (including test with the two versions of `Storage`) and ExtendedErrors
- Adjust iterator
- Add and test new PDElements functions
- Added `Bus.LineList` and `Bus.LoadList`
- Update test suite to include both `ExtendedErrors(False)` and `ExtendedErrors(True)`
- Testing running both debug and release builds of DSS C-API, using env var DSS_EXTENSIONS_DEBUG
- Added Python 3.8 to AppVeyor tests (Conda Python 3.8 on Windows was broken for awhile)
- Update version to 0.5.0

Closes #75, should help debug #84