dss-extensions/OpenDSSDirect.py #81: Unable to install OpenDSSDirect.py in Python 3.8.2

kdheepak, 2020-04-20:
The reason this issue occurs is because we don't have a OpenDSSDirect.py release that dumps the `dss_python` version. The version of `dss_python` in the latest pypi release is 0.10.1, which does not have a wheel for the latest versions of Python.

```
$ pip install OpenDSSDirect.py
Collecting OpenDSSDirect.py
  Using cached OpenDSSDirect.py-0.3.7.tar.gz (71 kB)
Collecting future
  Using cached future-0.18.2.tar.gz (829 kB)
Collecting six
  Using cached six-1.14.0-py2.py3-none-any.whl (10 kB)
ERROR: Could not find a version that satisfies the requirement dss_python==0.10.1 (from OpenDSSDirect.py) (from versions: 0.10.5)
ERROR: No matching distribution found for dss_python==0.10.1 (from OpenDSSDirect.py)
```

### Versions

- [ ] Python version: 3.8.2
- [ ] Python architecture: x64
- [ ] Operating system and version: MacOS

kdheepak, 2020-04-20:
A temporary workaround is to do something like this:

```
pip install "git+https://github.com/dss-extensions/OpenDSSDirect.py.git@master#egg=OpenDSSDirect.py[all]"
```

to install directly from the master branch.

PMeira, 2020-06-17:
New package (finally!) tested and uploaded, tagged as v0.4.0.

closed 2020-06-17 by PMeira