dss-extensions/OpenDSSDirect.py #84: Circuit.SetActiveBus() is not working

armandokeller, 2020-06-22:
### Versions

- [3.7.3] Python version: 
- [x64 ] Python architecture: 
- [ Windows 10] Operating system and version: 
- [0.4.0] OpenDSSDirect.py version number: 

### Bug

The method Circuit.SetActiveBus is not working properly. 

* **What is the current behavior?**
When the method Circuit.SetActiveBus(BusName) is called, it raises an exception expecting an Line Type and says that is using a Transformer.

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**
The active bus must be acessible by the name. I'm currently using the SetActiveBusi with a loop trough all busses, verifying the name of the bus to achieve this behavior.

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.**

``` python
import opendssdirect as dss
from opendssdirect.utils import *

run_command(f"Redirect {my_dss_file.dss}")

for bus in dss.Circuit.AllBusNames():
    print(bus)
    dss.Circuit.SetActiveBus(bus)
    # bus manipulation code
```

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)
I´m getting this error:

```
Traceback (most recent call last):
  File "***", line 35, in <module>
    dss.Circuit.SetActiveBus(bus)
  File "\lib\site-packages\opendssdirect\Circuit.py", line 88, in SetActiveBus
    return CheckForError(lib.Circuit_SetActiveBus(BusName))
  File "***\lib\site-packages\dss\_cffi_api_util.py", line 81, in CheckForError
    raise DssException(error_num, self._get_string(self._lib.Error_Get_Description()))
dss._cffi_api_util.DSSException: (5007, 'Line Type Expected, but another found. Dss Class=Transformer\r\nElement name=rtcam-p18_166') 
```
The element name on error message is a name of a transformer, not the bus name from the for loop.

PMeira, 2020-06-22:
Your code snippet by itself doesn't raise any issues for me, that's a clue.

This error is reported when you try to use most of the Line interface functions if the current active element is not a Line:

```
Line Type Expected, but another found. Dss Class=Transformer
Element name=rtcam-p18_166
```

Looks like you activate a transformer somehow and then use the Line interface. Since there is no error check for some of the Line methods, you get the error in the next setter, which seems to be `SetActiveBus`. OpenDSSDirect.py v0.4 maps the errors from the Error interface to exceptions automatically, in this case it's just that the first opportunity happens when you call `SetActiveBus`. We added this mostly for setters, but it seems we could extend it to many more functions.

The current API (based on the DSS C-API library) and the original one (the official OpenDSSDirect.DLL) both use the same behavior as the official COM API. So, in all implementations of OpenDSS, there's the concept of active element. For some components, like Lines, the global active circuit element is used. Some other components, like capacitors/transformers, use the active capacitor/transformer instead of the active circuit element.

PMeira, 2020-08-02:
@armandokeller Hi, did you find the root cause for the issue?

The new OpenDSSDirect.py release should help a lot for this kind of problem, for more details: https://github.com/dss-extensions/OpenDSSDirect.py/releases/tag/v0.5.0

PMeira, 2020-08-18:
Closing due to inactivity (we should probably get a bot for that). Please feel free to reopen if there's any remaining issue regarding this.

closed 2020-08-18 by PMeira