dss-extensions/OpenDSSDirect.py #86: Set Load not working

sanjaysrikumar, 2020-08-17:
### Versions

- [ ] Python version: 
- [ ] Python architecture: 
- [ ] Operating system and version: 
- [ ] OpenDSSDirect.py version number: 

Hello all, 

I am fairly a beginner to using this package as well as python. I have an OpenDSS circuit that i want to run by setting the load values for a few loads before proceeding to "Run" the simulation. 

My understanding is that using the `opendssdirect.Loads.Name(*args)` command will set the load "Active" 
and *args is the string name of my load in OpenDSS i.e. 'L1' from this piece of code in OpenDSS (new Load.L1 bus=B1814.1.2 Phases=1 kV=0.240 kW=1 kvar=0.484)

once i have set the specific load active, i use the following command to set a new value for its kW and kVar 
```
opendssdirect.Loads.Names('L1')
opendssdirect.Loads.kW=0.604
opendssdirect.Loads.kvar=0.311
```

This approach has not really worked for me and i also tried the approach of using opendssdirect.Loads.Idx() and i was not successful. 

Is this the right steps that i should be taking in order to change the load values ?

kdheepak, 2020-08-17:
I typically use the following:

```python
odd.Circuit.SetActiveClass("Load")
odd.Circuit.SetActiveElement("Load.L1")

odd.Loads.kW(0.604)
odd.Loads.kvar(0.311)
```

My understanding is that `odd.Loads.Name('L1')` will set the "current active load element"'s name to `"L1"`. But I think it does not change the `ActiveClass` to `Load`. So if the current `ActiveClass` is `Transformer` or `Line`, then I believe using this `odd.Loads.Name("L1")` won't do what you want it do it, and the following lines won't work because a `Load` object is not the `ActiveClass` or `ActiveElement`. Being explicit when using `opendssdirect` is helpful in avoiding these cases.

With the upcoming new API, some of these non-intuitive behaviors will likely be abstracted over so that it "just works".

PMeira, 2020-08-17:
@sanjaysrikumar To complement @kdheepak, OpenDSSDirect.py doesn't use (Python) properties, only functions so far -- docs here: https://dss-extensions.org/OpenDSSDirect.py/opendssdirect.html#module-opendssdirect.Loads

If you prefer properties for manipulating values or have legacy code written for the official COM API, you can use DSS_Python directly. DSS_Python and ODD.py share the same OpenDSS instance, you can use them together. 

In the long term, it's probably worth using properties in ODD.py too for simple fields like in this example, `Loads.kW` and `Loads.kvar`. But... in my opinion properties are overused in the OpenDSS API as a whole (both COM and internals). There's a lot to do before even considering doing this though.

See also: https://github.com/dss-extensions/OpenDSSDirect.py/issues/63#issuecomment-470627430

sanjaysrikumar, 2020-08-18:
Thank you both for your replies! 

I have a follow up question. So once i have dynamically changed or set the load values using the code: 
```
odd.Circuit.SetActiveClass("Load")
odd.Circuit.SetActiveElement("Load.L1")

odd.Loads.kW(0.604)
odd.Loads.kvar(0.311)
```
I now want to check if this worked so i want to subsequently store these new values in a new variable. 

for example: LoadkW=getLoadP . Here i am using getLoadP as an example of a code that i can use to retrieve these values using the OpenDSS package if such command exists... From my understanding odd.Loads.kW(0.604) "Sets" the kW to that specific value. I would like to know if a separate function can "Get" the previously Set value and store it into an array or something. 

Sanjay S

PMeira, 2020-08-18:
With very few exceptions, the functions in ODD.py that implement "setters" also expose "getters". So `odd.Loads.kW(0.604)` should set the value, and `odd.Loads.kW()` (without parameters) should return the current value. I guess we can add some notes about this convention to clarify it for new users.

PMeira, 2020-08-18:
Oh, to be clear, by "current" value I mean the specified/nominal value of the respective parameter. The actual operating/resulting value would of course depend on the system voltages.

sanjaysrikumar, 2020-09-17:
Hello all, 
I am back again with a question related to this very topic. 

I followed the instructions provided by you guys to "set" kW values to my load and it all seems to work well. Now, I am trying to see if the "get" also works and I've come across something strange. Maybe it's due to my lack of knowledge but nonetheless, I would like to clear it out.. 

So i have used the following piece of code to set a certain kW value to one of my loads. 
```
odd.Circuit.SetActiveClass("Load")
odd.Circuit.SetActiveElement("Load.L1")
odd.Loads.kW(5)

odd.Circuit.SetActiveClass("Load")  #Not sure if i need to "re-activate" this class since it was already activated previously. 
odd.Circuit.SetActiveElement("Load.L2")
odd.Loads.kW(10)
```

Now i wanted to test if i can retrieve these values back and store it by doing the following... 
`
odd.Circuit.SetActiveClass("Load")
odd.Circuit.SetActiveElement("Load.L1")
p1=odd.Loads.kW()
 `
p1 becomes 10 instead of 5. I have tried this with PVSystem to set PF and then later get them to verify if it worked well and I came across the same issue. Am i doing anything wrong? Please let me know what i can do eliminate this issue. 

Thank you!
Sanjay S

kdheepak, 2020-09-17:
> p1 becomes 10 instead of 5. 

If you set it to `10`, then it'll return `10`, right? Why do you expect or want it to return `5`?

If you call `odd.Loads.kW()` before setting it to `10`, it should return `5`.

sanjaysrikumar, 2020-09-17:
I have set Loads.L1 to have a kw of 5 and L2 to be 10. Later, i am attempting to retrieve the value set for L1 which is why I was expecting 5. 

If I understand correctly, if I am setting the kw of 10 different loads consecutively, I will not be able to retrieve the value of the first load since odd.Loads.kW() only stores the last value to be set?

kdheepak, 2020-09-17:
Ah, sorry. I misread your comment. Can you run `odd.Loads.Name()` before calling `odd.Loads.kW()` to verify that the load has been changed to the one you expect it to be?

sanjaysrikumar, 2020-09-17:
odd.Loads.Name() seems to return the last name of the loads. I have 43 loads and it returned L43 even though I have given the command to set the load L1. 

```
odd.Circuit.SetActiveClass("Load")
Out[77]: 20

odd.Circuit.SetActiveElement("Load.L1")
Out[78]: 176

odd.Loads.kW(7.5)

odd.Loads.Name()
Out[80]: 'l43'
```

kdheepak, 2020-09-18:
Can you try `odd.Circuit.SetActiveElement("L1")` instead of `odd.Circuit.SetActiveElement("Load.L1")`?

If that doesn't work try `odd.Circuit.SetActiveElement("l1")` or `odd.Circuit.SetActiveElement("load.l1")` (note the all lowercase names)?

PMeira, 2020-09-18:
@sanjaysrikumar, @kdheepak: I think I missed some details last time. 

Trying to clarify, each "module" represents a distinct interface for a set of components. Although they interact with each other (one interface might affect another), they can have distinct running states.
For loads, for example, they can same data/functions in the Loads module/interface, anything related to PCElements, anything related to CktElements, and finally general DSS objects. I think then the list would be:
1. Loads interface
2. Circuit + CktElement interfaces
3. ActiveClass + Property interfaces
4. DSS language commands (get and edit commands, dump, reports, etc.)

(Personally, I try to avoid 3 and 4 as much as possible since they are slow and error-prone. It's one of the reasons I started DSS C-API after all)

Each of these have different running states:
- With the exception of lines, each component-specific interface usually tracks its own active object (through its internal list). Usually only the specific interface of the circuit element affects or uses it.
- There is a global active circuit element (ActiveCktElement). This one is used through the CktElement interface but is affected by most interfaces of the specific circuit elements (e.g. Loads, Transformers, Lines -- all of them can affect the current CktElement).
- There is a global active DSS class (ActiveClass, or ActiveDSSClass internally). This one is used through the ActiveClass interface.
- There is a global active DSS object (ActiveDSSObject internally). This one is used through the Property interface.

These result in the limitation that you can only really have one active object of certain type at a given moment. We want to keep general compatibility with the official OpenDSS, so this concepts will remain as long as the official version follows them.

**1. Loads interface**

Calling `odd.Loads.Name('somename')`:
- will activate the load named `somename` in the Loads interface (or raise an exception if not found)
- will set the active circuit element. Most (probably all) interfaces for the individual component types behave this way. 
- **will not** update the ActiveClass
- will update ActiveDSSObject (*anything that changes the active circuit element does*)

So, if you intend to call the functions of the component's interface or related CktElement functions, this is enough. It's probably the preferred way since it uses some optimizations I've added starting on 2018.

**2. Circuit + CktElement interfaces**

Calling just `odd.Circuit.SetActiveElement("load.somename")` (yes, you should prepend the name with `load.` here):
- **will not** activate the element in the Loads interface
- will set the active circuit element. 
- will update ActiveClass 
- will update ActiveDSSObject

If you are using specific interfaces, such as `odd.Loads`, it won't work by itself, you need to set the active load through that interface.
If you are using general CktElement functions, this would be enough but prefer the specific interfaces if available (`odd.Loads.Name` et al). 
If you are using the ActiveClass or Property interfaces **for circuit elements**, this would be enough.

**3. ActiveClass + Property interfaces**

If you need to access something that is not exposed through a dedicated interface or is not a circuit element, you can use this alternative.

You have to activate the DSS class and the element itself, something like:

```python
odd.Circuit.SetActiveClass('load') # or odd.Basic.SetActiveClass('load'), same thing
odd.ActiveClass.Name('somename')
```

For loads, this idiom:
- **will not** activate the element in the Loads interface
- will set the active circuit element
- will update ActiveClass 
- will update ActiveDSSObject

---

**In conclusion**, it's probably better to use `odd.Loads.Name` and, if you use the ActiveClass interface, call `odd.Circuit.SetActiveClass("load")` to complement it.

If you need performance for large circuits, I'd recommend using the idx functions (e.g. https://dss-extensions.org/OpenDSSDirect.py/opendssdirect.html#opendssdirect.Loads.Idx ), coupled with `odd.Circuit.SetActiveClass` if required:
- no strings are encoded/decoded through the APIs (both CPython's C-API and DSS C-API), just a simple integer
- trivial/fast at Pascal level since it's a basic array access and basic checks. No string copies, no lowercase conversions, no hashes.

---

**Sidenotes:** 

1. It might be worth checking if updating the ActiveClass on `odd.Loads.Name('somename')` at Pascal level (on DSS C-API) breaks any legacy code. The same for `odd.Circuit.SetActiveElement("load.somename")` and updating the current active load, etc. Maybe open a related ticket on DSS C-API?

2. In the medium/long term, we can use a different implementation that simplifies all of these in the API/end-user side, while still providing the old/legacy API for compatibility at user-level for old code. We could make all these `Active*` objects optional, making the references to the objects unique, and reimplement the current/legacy API using the new concepts.

E.g. in the future, at Python level, we'll probably have something like:

```python
load1 = odd.Loads['somename']
load2 = odd.Loads['othername']
```

The `load1` and `load2` objects will expose the current Loads interface, while inheriting the CktElement and Property interfaces, besides providing other extensions. It will be possible to read/write to both objects without worries about which one is active on the OpenDSS side.

At Pascal level, there is already inheritance and all these interfaces are exposed in a single object -- I'm not sure why this wasn't better exploited in the COM API (both the official OpenDSSDirect.DLL and our DSS C-API were initially derived from the COM implementation).

samtalki, 2020-12-09:
Not sure if this is helpful or not but I think something along the lines of this might be able to accomplish your goal, OP, by interfacing directly with OpenDSS. Please disregard if this isn't what you're looking for.

`def placeInjection(bus_name,phases,kw_inj,kvar_inj):
    dss.run_command(
        "New Load.{bus_name} Bus1={bus_name} phases={phases} kW ={kw_inj} kvar={kvar_inj}".format(
        bus_name=bus_name,
        phases = phases,
        kw_inj = kw_inj,
        kvar_inj = kvar_inj
        ))
`

PMeira, 2022-07-16:
Moving this to dss-extensions since it's a good discussion about the current organization. Now that the essential work is done, this should finally be addressed after some planning.