dss-extensions/OpenDSSDirect.py #89: Exprot to file?

iboates, 2020-10-20:
Is there a way to save to a DSS file? I don't see any "save" or "export" methods in the code

kdheepak, 2020-10-20:
What would you like to export? The DSS model? I don't think that is possible with this package. 

You can however write out data after reading from every attribute in the `.dss` format. Check out https://github.com/NREL/ditto/ for that.

PMeira, 2020-10-21:
Indeed there is no explicit API functions (I added a new issue in DSS C-API to remember that: https://github.com/dss-extensions/dss_capi/issues/88), but you can use DSS commands to save the circuit, whole classes of elements, or single elements. There are some corner cases where it doesn't work perfectly (e.g. complex LineGeometry elements) so it's best to check to ensure the output is OK.

Excerpts from the [OpenDSS Manual](https://sourceforge.net/p/electricdss/code/HEAD/tree/trunk/Version8/Distrib/Doc/OpenDSSManual.pdf) below.

> **Save**
> Syntax:
>     ` [class=]{Meters | Circuit | Voltages | (classname)} [file=]filename [dir=]directory` 
> 
> Default class = Meters, which saves the present values in both monitors and energy meters in the active circuit. 
> **"Save Circuit"** saves the present enabled circuit elements to the specified subdirectory in standard DSS  form  with  a  Master.txt  file  and  separate  files  for  each  class  of  data.  If  Dir=  not  specified  a unique name based on the circuit name is created automatically.  If Dir= is specified, any existing files are overwritten.
> 
> **"Save Voltages"** saves the present solution in a simple CSV format in a text file. Used for VDIFF command.
> 
> Any class can be saved to a file.  If no filename is specified, the classname is used. 
> 
> The filename of the file written appears in the Result window after execution of this command.

(For usage from APIs, instead of the "Result window", the filename should be in the Result property/function of the Text interface.)

> **Dump  <Circuit Element> \[Debug\]**
> Writes a text file showing all the properties of the circuit object and displays it with the DSS text editor.  You would use this command to check the definition of elements.  It is also printed in a format  that  would  allow  it  to  be  fed  back  into  the  DSS  with  no,  or  only  minor,  editing.    If  the “dump” command is used without an object reference, all elements in the active circuit are dumped to a file, which could be quite voluminous.
> 
> If the Debug option is specified, considerably more information is dumped including the primitive Y matrices and other internal DSS information.
> 
> You can limit the dump to all elements of a specific class by using the wildcard, *, character: For example,
> 
>     Dump Transformer.* debug
> 
> Will dump all information on all transformers.Leave “debug” off to just see property values.
>

closed 2020-10-21 by PMeira