dss-extensions/OpenDSSDirect.py #93: Raise exceptions instead of returning error codes or strings

daniel-thom, 2021-02-03:
This is an idea for improving usage of OpenDSSDirect.py. Feel free to close it if you judge it non-ideal or impractical.

Problem: Many users do not check the return value from OpenDSSDirect function calls. You can certainly argue that they should, and I do that. But they don't. They may call something like `SetActiveBus` with an invalid name, not check that -1 was returned and then continue with an incorrect assumption.

One reason users don't check for these errors is that it makes the code extremely verbose, especially when they are making lots of calls.

Solution: OpenDSSDirect.py could raise an exception any time an error is encountered. It would simplify scripts and make them more robust. In almost all cases I think that the right thing to do is abort execution (at least in the cases I'm considering). I've been tempted to make wrapper functions to achieve the same result.

A byproduct would be unifying the returned errors (`run_command` returns strings, others return integers).

I suppose the behavior could be configurable in order to avoid breaking existing code.

There are plenty of reasons not to do this; just raising it for consideration.

PMeira, 2021-02-03:
> Solution: OpenDSSDirect.py could raise an exception any time an error is encountered. 

Unless I missed something, this is already something in progress... `SetActiveBus`, for example, does check if an error is marked by the OpenDSS engine and raises an exception: 

https://github.com/dss-extensions/OpenDSSDirect.py/blob/3b553d137903127a9fb838c5d37927ac7e303497/opendssdirect/Circuit.py#L87

I imagine your issue might be that OpenDSS does not consider somethings errors (silently ignores them or returns numeric flag values): 

https://github.com/dss-extensions/dss_capi/blob/a3ec9a105c36d74f9ff5146f28862f714c76dcf9/src/v7/CAPI_Circuit.pas#L630-L637

That's another issue and should be handled in our OpenDSS engine since it's shared with the other extensions/language bindings. A lot of extra error checking has been added in there, especially at the API layer level. I'd recommend checking DSS C-API commits, changelogs, issues for detailed context. There is a lot of things planned still. 
The current error interface is used by most DSS Extensions to map to exceptions, even DSS_MATLAB does that. For the SetActiveBus issue, that seems to fit the `DSS_CAPI_EXTENDED_ERRORS` approach, so we could add another `if` to raise an error instead of returning -1 if the mechanism is enabled. We could open another ticket to collect candidate functions for this kind of treatment.

Note that it is essential to keep good compatibility with the official OpenDSS, we cannot just rewrite everything. A lot of the changes so far can be toggled to restore a behavior closed to EPRI's distribution. Usually I gather information on one type of issue, scan the whole API (if not the whole Pascal codebase), and plan changes in batches to avoid tiny commits related to sparse changes. 

> A byproduct would be unifying the returned errors.

In the long term, we should to rewrite the whole thing. Right now, I imagine for the next 6 months or so, the focus is keeping things somewhat compatible for historical reasons, both here in ODD.py and DSS Python, while some other features are added in DSS C-API.

> run_command returns strings

#70 

> others return integers

There is also one or two instances where a tuple is returned. We ported most of these from the COM implementation (like DSS C-API was created initially, the official OpenDSSDirect API is also derived from the COM interfaces). If we were to design an API from scratch, it would be more consistent.

A new design would be welcomed after DSS C-API 0.11 or 0.12 is released. By the current plans, the 0.12 series should be the last one before a full 1.0 release. We'll probably keep the current interfaces for a while, but another one can be added in parallel.

> There are plenty of reasons not to do this; just raising it for consideration.

From my point-of-view, as long as performance is not affected much, the earlier the error is reported, the better. We can move some things down to DSS Python in Cython if performance becomes an issue -- currently we use CFFI but Cython was also tested before.

daniel-thom, 2021-02-03:
Thanks for the detailed response. The main issue is that I'm currently stuck on v0.3.7 and I forgot that you added a lot of error checking in later versions. I will encourage the people I'm dependent on to upgrade. Afterwards, I'll dig into the specifics of any situation I encounter. Closing.

closed 2021-02-03 by daniel-thom

PMeira, 2021-02-04:
@daniel-thom Ah, I see. And yes, v0.3.7 is almost 2 years old.
I created https://github.com/dss-extensions/dss_capi/issues/91 so we don't forget to add better handling in `SetActiveBus` etc.