dss-extensions/OpenDSSDirect.py #94: Question: Diference between imports

felipemarkson, 2021-02-21:
### Versions

- [x] Python version: v3.8.6
- [x] Python architecture: x64
- [x] Operating system and version: Ubuntu 20.04.1 LTS
- [x] OpenDSSDirect.py version number: v0.6.0

### What is the difference between these imports?
I've tried to encapsulate the OpenDSSDirect.py in a Class to use with parallelism. Thus, I've tried these three imports and I don't see any difference between them.  

```python
import opendssdirect
```

```python
from opendssdirect import dss
```

```python
from opendssdirect.dss import dss
```

Is there a way to create another independent instance of dss that doesn't share the circuits or data?

kdheepak, 2021-02-21:
There's no difference between these imports (except for Python scoping, and even then only the first will behave differently). 

In terms of parallelization, I think you'll have to use `dask.distributed` or `multiprocessing` in Python and `import opendssdirect` in each worker or subprocess. If you try to use threading, the Python process will share the same instance of opendss engine. Additionally, I believe you will have to `import opendssdirect` within the worker process, and not before you fork in the main process. 

I've used this successfully with `dask.distributed`. If you spawn a worker with dask, it starts a new process so there's no sharing of memory. I believe the same will happen for `multiprocessing` but I haven't tested that.

felipemarkson, 2021-02-21:
Thanks for your answer!

Is there a way to create different instances of ``opendssdirect`` or ``dss-python`` with memory independence in the same thread?

I think it is useful to simulate differents circuits in the same thread, and can avoid some memory issues.

PMeira, 2021-02-22:
> Is there a way to create different instances of opendssdirect or dss-python with memory independence in the same thread?

Not yet, but soon.

felipemarkson, 2021-02-24:
Thanks, @PMeira.

I'm going to close this issue and wait for this update.

closed 2021-02-24 by felipemarkson