dss-extensions/OpenDSSDirect.py #100: DSSException

Akeruyri, 2021-06-18:
- [ ] Python version: 3.8
- [ ] Python architecture: x86
- [ ] Operating system and version: Win 10
- [ ] OpenDSSDirect.py version number: 0.6.1

### Support Question

Perhaps this may be better asked on the OpenDSS Forums, but anyways...

I am writing a class that can import the voltage regulators (RegControl Objects) from a given .dss file and run tap changes on all of them. When testing this class with the IEEE Test Cases it pulls regulators in the IEEE 13 Bus and IEEE 123 Bus systems just fine, but trying the 8500 node system gives this error. 

[image]
dss._cffi_api_util.DSSException: (485, 'Warning Max Control Iterations Exceeded. \r\nTip: Show Eventlog to debug control settings.')

Im guessing it has to do with some of the regulators on that system not having 33 taps (+- 16 and 0) but if its a more complex issue than that I wouldn't know where to start in fixing it.  It does say the command failing is the RegControls_Set_Name, it may also be an issue with how im importing all the RegControl Objects, but the original line that is failing is the switch_taps function

Below is the code its running
   ```

###In the Test file###
#Load in the model
dss.run_command(r'compile "C:\Users\louis\Desktop\SeniorDesignProject\OpenDSS\8500-Node\Master.dss"')

#Get list of Regulators

regulator_list = dss.RegControls.AllNames()
regulators = []
for name in regulator_list:
    regulators.append(reg_controller(name))

#Run Tap Change Test
for hrs in range(regulators[0].total_timesteps):
    for reg in range(len(regulators)):
        regulators[reg].switch_taps(hrs) #Change the taps
    dss.run_command('solve')

###In the Class###
    def __init__(self, name):
        self.reg_name = name

        # When starting up, the controller needs access to information from the circuit
        dss.LoadShape.First()  # Sets first loadshape as active loadshape
        self.total_timesteps = dss.LoadShape.Npts()  # A tap change may occur on each loadshape interval

        self.tap_scale = dss.LoadShape.HrInterval()  # Returns the interval between each loadshape points in hrs
        self.total_time = self.tap_scale * self.total_timesteps  # Length of the loadshape in hours
        self.tap_list = np.zeros((self.total_timesteps,),dtype=int)  # Store a 2D array that will hold all of our tap changes for each SVR

        #Create Tap List from Loadshape
        self.loadshape_to_tap() # Populate the tap_list and other values

    def loadshape_to_tap(self): #Simple Implementation, A better method can be developed later
        self.tap_scale = dss.LoadShape.HrInterval()  # Returns the interval between each loadshape points in hrs
        self.total_time = self.tap_scale * self.total_timesteps  # Length of the loadshape in hours
        mult_unscaled = dss.LoadShape.PMult() #pull list of multipliers
        multipliers = np.interp(mult_unscaled, (min(mult_unscaled),max(mult_unscaled)),(-16,16)) #Range them to the taps.
        for each in range(multipliers.size):
            multipliers[each] = math.trunc(multipliers[each])  # Truncate the tap values
            self.tap_list[each] = multipliers[each] # Assign the taps. All SVRs will run off the same tap for now.

    def switch_taps(self, timestep):
        dss.RegControls.Name(self.reg_name) #Set active SVR
        dss.RegControls.TapNumber(self.tap_list[timestep]) #Attempt a tap change.

```
Thanks!

PMeira, 2021-06-23:
Try replacing `dss.run_command` with `dss.Text.Command`. The former doesn't check for errors, so the exception is delayed until something else does (in this case, when you set the regcontrol name).

On the convergence issue per se, if the system did not converge, there may not be anything to "fix". You can increase control iterations and so on, but usually if there's so many control iterations without convergence, you may need to rethink the control logic.

Akeruyri, 2021-06-23:
Ok I will try that, thank you

closed 2021-06-23 by Akeruyri