dss-extensions/OpenDSSDirect.py #102: Documentation of Irradiance units in PVsystem module different from OpenDSS

tarekelgindy, 2021-06-22:
### Versions

- [ ] Python version: 3.7
- [ ] Python architecture: x64
- [ ] Operating system and version: CentOS-7
- [ ] OpenDSSDirect.py version number: v0.3.7

### Support Question

I've noticed that there are differences in the documentation between OpenDSSDirect.py and OpenDSS for the Irradiance Property of the PVsystems module:

https://dss-extensions.org/OpenDSSDirect.py/opendssdirect.html#opendssdirect.PVsystems.Irradiance

The units here say W/m<sup>2</sup> but in the OpenDSS help, indicate that it's in kW/m<sup>2</sup> in version 8.5.9.1:

[image]

After setting the irradiance in OpenDSSDirect.py the behavior seemed to be more sensible when I used units for kW/m<sup>2</sup> so I suspect this is just a documentation issue.

The code I used to check it (after loading the data into memory but not solving it) looked like this:

```python

# Already loaded model
first_pv = dss.PVsystems.First()
while True and first_pv > 0:
    name = dss.PVsystems.Name()
    irradiance = pv_profile_data[name]
    res1 = dss.PVsystems.Irradiance(irradiance/1000)
    if not dss.PVsystems.Next() > 0:
        break

print('starting opendss run')
result = dss.run_command(f"Solve")

print('finished opendss run')
print(result)

dss.Circuit.SetActiveClass("PVsystem")
flag = dss.ActiveClass.First()
while flag>0:
    print(dss.CktElement.Name(), dss.CktElement.Powers())
    flag = dss.ActiveClass.Next()

```

The powers were at the KVA limit for the PVsystem object when I wasn't dividing by 1000 which indicated to me that the irradiance was too high.

Based on this line, it should be calling `self._lib.PVSystems_Set_Irradiance(Value)`:

https://github.com/dss-extensions/dss_python/blob/18aa20442a522372fff7095ff874aaca0facc149/dss/dss_capi_gr/IPVSystems.py#L40

PMeira, 2021-06-23:
As a side note, since you're comparing with OpenDSS 8.x, I assume enabled the legacy models, right? Otherwise, we default to the 9+ version which is a different implementation.

Those in the EXE are the "DSS properties", ours are listed in https://github.com/dss-extensions/dss_capi/blob/master/docs/dss_properties.md#pvsystem-properties (for 9+ models).

The API-level properties mirror the COM API, including the docs:

https://github.com/dss-extensions/dss_capi/blob/80a462fc0028e7dbd1e1a4851520024a42423065/Version8/Source/DLL/OpenDSSengine.ridl#L2818-L2821

The official code returns the internal value...

https://github.com/dss-extensions/dss_capi/blob/80a462fc0028e7dbd1e1a4851520024a42423065/Version8/Source/DLL/ImplPVSystems.pas#L250

...and so does our version:

https://github.com/dss-extensions/dss_capi/blob/62e3daedced2d26118bae57f2a9d568d3b7905ff/src/CAPI/CAPI_PVSystems.pas#L295

(The same is valid for `IrradianceNow`)

Now, the actual code of the component indicates it is indeed kW/m² (and I didn't find anything scaling it by 1/1000 either):

https://github.com/dss-extensions/dss_capi/blob/62e3daedced2d26118bae57f2a9d568d3b7905ff/src/PCElements/PVsystem2.pas#L1094

So, you're right. We should probably just add the missing "k" to the API docstrings.

Besides the .RIDL file, [the official CHM docs](https://sourceforge.net/p/electricdss/code/HEAD/tree/trunk/Version8/Distrib/x64/OpenDSS_COM.chm) also list W/m², missing the k:

[image]

PMeira, 2021-06-23:
> As a side note, since you're comparing with OpenDSS 8.x, I assume enabled the legacy models, right? Otherwise, we default to the 9+ version which is a different implementation.

Oops, nevermind that for older versions:

> OpenDSSDirect.py version number: v0.3.7

ODD.py v0.3.7 forces DSS Python 0.10.1, which is quite old.

```python
install_requires=["future", "six", "dss_python==0.10.1"],
```

tarekelgindy, 2021-06-25:
Ahh yes, I think I just ran this with the version I happened to have locally. I should probably update to to be 9.x

Good to know this is probably just a typo type error! I'll proceed with the irradiance being in kW/m<sup>2</sup>

closed 2022-04-21 by PMeira