dss-extensions/OpenDSSDirect.py #106: Accessing monitor data in memory (without writing to a textfile)?

eddie-atkinson, 2022-01-24:
### Versions

- [x] Python version: 3.9.7
- [x] Python architecture: x86
- [x] Operating system and version: Mac OS Catalina 10.15.7-->
- [x] OpenDSSDirect.py version number: 0.6.1

### Support Question

I am new to OpenDSS and am interested in using it to model an LV network for the purposes of training reinforcement learning agents. One thing that I have been unable to do in my testing is to access data from monitors in memory without first writing them to disk.

In my experience interfacing with the file system typically results in a high runtime penalty which would will be unfortunate when I get to training models which may require thousands of epochs to learn. 

Others who are more experienced with OpenDSS than me have said that the ability to access monitor data from memory is exposed by a lower level API (C, I believe). Is this functionality exposed in this package? If not, does a lower level API exist that I may be able to interact with to build the required functionality? Of course I would be happy to open a PR with my solution if it is of interest to others.

PMeira, 2022-01-26:
> In my experience interfacing with the file system typically results in a high runtime penalty which would will be unfortunate when I get to training models which may require thousands of epochs to learn.

You don't need to write monitor data to disk. I guess that writing to disk in this case would be unusual, unless the user is running through the main OpenDSS GUI.
Usually everything (and a bit more) from the official COM API is available on DSS Extensions. 

For OpenDSSDirect.py, the relevant API is documented here:
https://dss-extensions.org/OpenDSSDirect.py/opendssdirect.html#module-opendssdirect.Monitors

Most of the functions from [DSS C-API](https://github.com/dss-extensions/dss_capi/blob/master/include/dss_capi.h) are exposed in Python.

Both OpenDSSDirect.py and DSS Python have the extra `AsMatrix` function that returns all monitor data as a NumPy matrix.
OpenDSSDirect.py also has a `monitor_to_dataframe` function, although manually creating a dataframe is not hard as seen below.

An example:
```python
from opendssdirect import dss
from matplotlib import pyplot as plt

# Load the basic IEEE13 circuit (adjust the path before running)
dss.Text.Command('redirect "IEEETestCases/13Bus/IEEE13Nodeckt.dss"')

# Add some changes and a monitor
dss.Text.Command('batchedit Load..* Daily=Default')
dss.Text.Command('new Monitor.TestMon element=Load.675a')

# Solve for a day, 15-min steps
dss.Text.Command('solve mode=Daily stepsize=900 number=96')

# Activate the first monitor and inspect the names
dss.Monitors.First()
print(dss.Monitors.Name())
header = dss.Monitors.Header()
print(header)

# If you need just one channel, you can use the Channel and dblHour functions.
# These are the same as the Channel/dblHour functions from the COM API and DSS Python.
ch = 1
single_channel = dss.Monitors.Channel(ch)
hours = dss.Monitors.dblHour()
plt.figure()
plt.plot(hours, single_channel)
plt.xlabel('Time (h)')
plt.ylabel(header[ch - 1].strip())

# Depending on the use-case, reading all channels via AsMatrix is faster
# AsMatrix is only available in DSS Extensions (OpenDSSDirect.py, DSS Python, etc.)
full_mon_data = dss.Monitors.AsMatrix()

# Example plot
hours = full_mon_data[:, 0] + full_mon_data[:, 1] / 3600.0
plt.figure()
plt.plot(hours, full_mon_data[:, 1 + ch])
plt.xlabel('Time (h)')
plt.ylabel(header[ch - 1].strip())
plt.show()

# For interactive inspection, you could also convert to a dataframe
# Using "monitor_to_dataframe"...
from opendssdirect.utils import monitor_to_dataframe
df = monitor_to_dataframe()
df = df.set_index(df.hour + df.second / 3600)
df.V1.plot()

#...or manually assembling the dataframe
import pandas as pd
full_header = ['H', 'S'] + [name.strip() for name in header]
df = pd.DataFrame(full_mon_data, columns=full_header)
df['time'] = df.eval('H + S / 3600')
del df['H'], df['S']
df = df.set_index('time')
df.V1.plot()
```

@eddie-atkinson Is this enough to solve your issue?

eddie-atkinson, 2022-01-27:
That's perfect. Thank you so much @PMeira  :)

closed 2022-01-27 by eddie-atkinson