dss-extensions/OpenDSSDirect.py #108: OpenDSS internal export path issue

AadilLatif, 2022-02-18:
When I export OpenDSS internal reports, they seem to be ending in the code directory rather than the model directory.

**Python Code** 

dss_folder = os.path.dirname(new_model_path)
os.chdir(dss_folder)
print(f"working _directory {os.getcwd()}")

self.dss_instance.run_command('Clear')
message = self.dss_instance.run_command(f"Redirect Master_wls.dss")

**OpenDSS model**

Clear
Redirect loadshape.dss 

New Circuit.Full_Network bus1=7546407 pu=1.0 basekV=6.9 r1=0.04 x1=0.523 r0=0.023 x0=0.427 

Redirect LineCodes.dss
Redirect Lines.dss
Redirect Transformers.dss
Redirect Loads.dss
Redirect updates.dss 

Set Voltagebases=[0.24000704030977513, 0.4156921938165305, 6.9]

Calcvoltagebases

Buscoords Buscoords.dss

Set CaseName=PRECISE
set mode=yearly 

Set overloadreport=true  ! TURN OVERLOAD REPORT ON
Set voltexcept=true
set demand=true
set DIVerbose=true

Set Year=0 Number= 4796
solve 
closedi

PMeira, 2022-02-18:
Hi, which version you're using?

kdheepak, 2022-02-18:
I think @AadilLatif is using an extremely old version because of the wheel issue that I had posted an issue about a while ago. But iirc that’s been resolved. So @AadilLatif, you should also update to the latest version of OpenDSSDirect. It’ll help you catch other issues with your model too.

PMeira, 2022-02-25:
@AadilLatif I agree with Dheepak's comments, you may have installed in the window that Python 3.10 wheels were not available. You may need to reset your local cache if it persists (or install with `pip install --no-cache ...`). I'll close the issue but please feel free to reopen in case it's something else.

@kdheepak I don't think this will be a problem in the future. In case it does happen again, we could remove the wheels from PyPI and archive then here and somewhere else. DSS C-API is mirrored on SourceForge, including the releases: https://sourceforge.net/projects/dss-capi/files/

closed 2022-02-25 by PMeira