dss-extensions/OpenDSSDirect.py #109: dss..CktElement.Enabled() issue

GSkrt, 2022-03-06:
### Versions

- [3.8 ] Python version: 
- [ x64] Python architecture: 
- [ Win 10] Operating system and version: 
- [0.6.1 ] OpenDSSDirect.py version number: 

### Bug
Not sure if its a bug but...
I want to have some control over regulators in my model. I'm using 

'''        isin = dss.RegControls.First()
        tap_settings = {}
        while isin > 0:
             ime = dss.RegControls.Name()
            tap = dss.RegControls.TapNumber()
            transformer = dss.RegControls.Transformer()
            tap_settings[ime] = (transformer, tap)
            dss.CktElement.Enabled(enable_disable) # I want to enable or disable regulator
            isin = dss.RegControls.Next() '''

Second loop when I want to enable regulators again gives me: 

` isin = dss.RegControls.First() ==0'

Is this a bug or I don't understand something here ?  I was expecting regulators to be there in the next loop just not enabled. 

Thanks for your help.

PMeira, 2022-03-06:
Not a bug, OpenDSS skips disabled circuit elements with `First()`/`Next()`. `Count()` still includes everything, so it can be a little surprising.

You can still switch the element by `Name()` or `Idx()`. In DSS Extensions, all elements expose the index through `Idx()`, so I'd recommend using that whenever possible (it's fast): https://dss-extensions.org/OpenDSSDirect.py/search.html?q=Idx

One other surprising behavior is that most `First()`/`Next()` return the actual `Idx()`, while some others (Monitors, PDElements) use just 1 or 0.

We'll probably won't change the default behavior for `First()`/`Next()` iteration to keep some compatibility with EPRI's version. Still, next version will expose a [new toggle from DSS C-API](https://github.com/dss-extensions/dss_capi/blob/ccefd69b06dfe09af5daa2d31da7967f14275511/src/CAPI/CAPI_Settings.pas#L51-L52) to include disabled elements.

GSkrt, 2022-03-07:
Thank you for your reply and solution. 😃 ill try this afternoon to see how it works.

closed 2022-04-13 by PMeira