dss-extensions/OpenDSSDirect.py #115: Incidence Matrix

keithmoffat, 2022-07-07:
### Versions

- [ ] Python version: 3.7.11
- [ ] Python architecture: x64
- [ ] Operating system and version: OSX 11
- [ ] OpenDSSDirect.py version number: 0.6.1

### Feature Request

Would it be possible to get the incidence matrix for a network in opendssdirect?

I have tried dss.Solution.IncMatrix(), but only got [0]. Maybe I am using it wrong..

It looks like dss has CalcincMatrix(), CalcincMatrix_O(), and Export IncMatrix(), so perhaps it would be an easy add.

PMeira, 2022-07-07:
> It looks like dss has CalcincMatrix(), CalcincMatrix_O(), and Export IncMatrix(), so perhaps it would be an easy add.

@keithmoffat  Those three are commands, they're already implemented here. Here's an example:

```python
import numpy as np
from scipy.sparse import coo_matrix
from matplotlib import pyplot as plt
import opendssdirect as odd

odd.Text.Command('redirect EPRITestCircuits/ckt5/Master_ckt5.dss')
odd.Text.Command('calcincmatrix')
data = odd.Solution.IncMatrix()[:-1]
x, y, v = np.asarray(data[0::3]), np.asarray(np.asarray(data[1::3])), data[2::3]
inc_mat = coo_matrix((v, (x, y)))
plt.spy(inc_mat)
```

I agree that's not great but that's how it was implemented in OpenDSS. We should expose the matrix as a coo_matrix directly here, but at the same time it's best to not deviate too much. @kdheepak , it might be better for us to add a parallel modern API and keep DSS_Python and ODD.py as-is for backwards compatiblity.

kdheepak, 2022-07-07:
I completely agree about a parallel effort to create a better API.

closed 2022-07-16 by PMeira