dss-extensions/OpenDSSDirect.py #117: OpenDSS compile unknown parameter error: VV_RefReactivePower parameter for InvControl Module

gayukrishna, 2022-08-05:
### Versions

- Python version: 3.7
- Operating system: Linux
- OpenDSSDirect.py version number: v0.7.0

### Bug

 OpenDSSdirect.py (v0.7.0) doesn't recognize the VV_RefReactivePower parameter in the inverter control module

To reproduce: Complie any opendss master script with InvControl module using VV_RefReactivePower in versions defined above
[image]

PMeira, 2022-08-05:
That option was marked deprecated years ago in OpenDSS v9.0 when the model for InvControl was fully replaced:

https://sourceforge.net/p/electricdss/code/HEAD/tree/trunk/Version8/Source/Controls/InvControl.pas#l691

We removed some of the deprecated features in our latest release.

We could readd the deprecated parameters just include notes such as *"Use RefReactivePower instead"* the error message, but the point of deprecation is to eventually remove. I noticed a lot of users didn't even realize the models for InvControl, Inverter, Storage, StorageControl were replaced yet.

Also note that all example circuits from EPRI replaced `VV_RefReactivePower` with `RefReactivePower` when the parameter was deprecated.

PMeira, 2022-08-05:
I added a mechanism for providing better messages for removed items, in this case it would result in:

> DSSException: (#2020030) InvControl.testic.VV_RefReactivePower: "VV_RefReactivePower" was deprecated in 2020. Use "RefReactivePower" instead.

This will be included in the next version coming in a week or two. I'll leave this ticket open to see if we'll get any feedback.

To illustrate why removing is better than keeping things indefinitely, see for example this: https://sourceforge.net/p/electricdss/discussion/861976/thread/658e9da071/ -- the user was hopelessly changing `%idlingkvar` for the current Storage model and expecting changes in behavior. This happens quite a lot, especially since PVSystem and Storage models are used in popular research topics.

But this issue also highlights that we need a warning system so we can warn users before removing old parameters, besides other script issues. For that, I opened https://github.com/dss-extensions/dss-extensions/issues/16 for general feedback.

PMeira, 2022-08-05:
Another link from EPRI recommending to replace `VV_RefReactivePower`: https://sourceforge.net/p/electricdss/discussion/experts/thread/6f17e5f375/?limit=25#2ed8

PMeira, 2022-08-10:
Another one that was removed -- `kvarlimit` in PVSystem: https://sourceforge.net/p/electricdss/discussion/861976/thread/4a6aa2199f/

This one was already removed in the official version, so users will need to update their scripts anyway. By the way, the error message reported there is from DSS Extensions -- we added the traceback such as `[file: "PVSystems.dss", line: 255]`

PMeira, 2022-08-10:
For reference, here's the doc for the "new" models:

https://sourceforge.net/p/electricdss/code/HEAD/tree/trunk/Version8/Distrib/Doc/PVSystem_InvControl.pdf?format=raw

There seems to be some confusion regarding these.

PMeira, 2022-08-12:
New info posted on the forum: https://sourceforge.net/p/electricdss/discussion/861977/thread/25a9f092bf/?limit=25#e929

The deprecated parameter `PVSystemList` had a bug, both here and in the official version.

closed 2022-12-01 by PMeira