dss-extensions/OpenDSSDirect.py #120: Get power flow result

goruiqi, 2022-12-02:
Hello all. I want to get the power flow result,  such as power consumed by load and power flowing through the line. How can I get them? Looking for your reply.

tshort, 2022-12-02:
You have several options. See:

https://dss-extensions.org/OpenDSSDirect.py/opendssdirect.html#module-opendssdirect.CktElement

PMeira, 2022-12-03:
@goruiqi The docs @tshort linked should give what you need. If you're new to OpenDSS, I recommend checking the official documentation from EPRI, either from a local OpenDSS installation, or some folders from the official SVN repo like https://sourceforge.net/p/electricdss/code/HEAD/tree/trunk/Version8/Distrib/Doc/ and https://sourceforge.net/p/electricdss/code/HEAD/tree/trunk/Training/

To help a bit, I uploaded some notes, including a table that maps the COM (and DSS-Python) functions/properties to the ODD.py API: https://github.com/dss-extensions/dss-extensions/blob/main/python_apis.md. As with the rests of the docs, this will be expanded in the future, but you can quickly see the pattern there. More links at https://dss-extensions.org/#documentation

closed 2022-12-03 by PMeira