dss-extensions/OpenDSSDirect.py #121: Error 303 Reported From OpenDSS Intrinsic Function

KapilDuwadi, 2023-02-07:
### Versions

- [ x] Python version: 3.9
- [ x] Python architecture: x64
- [ x] Operating system and version: Windows 10
- [x ] OpenDSSDirect.py version number: 0.7.0

### Feature Request

### Bug

* **What is the current behavior?***
Running into issue executing the following command.
`dss.Text.Command('export cim100')`

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**

I was expecting a xml file output. The command works in OpenDSS GUI.

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.*** 

You can use any opendss model and run the command `dss.Text.Command('export cim100')` to reproduce the bug.

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

Traceback (most recent call last):
  File "opendss_xml\main.py", line 30, in <module>
    generate_xml_file()
  File "opendss_xml\main.py", line 28, in generate_xml_file
    dss.Text.Command("Export CIM100")
  File "opendssdirect\Text.py", line 17, in Command
    CheckForError()
  File "\lib\site-packages\dss\_cffi_api_util.py", line 128, in _check_for_error
    raise DSSException(error_num, self._get_string(self._lib.Error_Get_Description()))
dss._cffi_api_util.DSSException: (#303) Error 303 Reported From OpenDSS Intrinsic Function:

PMeira, 2023-02-07:
Thanks for the report. There are some pending changes from the upstream OpenDSS that could help. I'll also try to add a test so it doesn't break again.

We should have a fixed release by next Monday.

PMeira, 2023-03-28:
Done and new packages uploaded to PyPI. We'll prepare a release message on GitHub soon too, for the moment see the relevant changes at https://github.com/dss-extensions/dss_python/blob/master/docs/changelog.md#0130

Sorry for the delay, although this issue was easy to fix, some other issues needed more time.

On CIM specifically, I had to compare with OpenDSSCmd (which is also non-official, even though it's hosted besides the main OpenDSS) since there hasn't been an official OpenDSS release after several changes to the XML export. I have some reservations about OpenDSSCmd, but at least the output matched exactly.

I believe that, long term, it would be beneficial to move this code to a plug-in. Right now, there are a lot of intrusive and unnecessary changes just for CIMXML. Maybe [DiTTo](https://github.com/NREL/ditto/) or a similar project would be a better home. I added some basic tests but such feature really needs a more extensive test-suite. Maybe there's already a test-suite outside of the OpenDSS SVN repository, but I couldn't find it yet.

closed 2023-03-28 by PMeira