dss-extensions/OpenDSSDirect.py #122: opendssdirect crashes with bad linecode

wenbowangnrel, 2023-03-09:
### Versions

- [ ] Python version:  3.10
- [ ] Python architecture: x64
- [ ] Operating system and version: Centos 7
- [ ] OpenDSSDirect.py version number:  0.7.0

### Bug

* **What is the current behavior?** I can compile and solve a circuit with a bad linecode in opendssdirect.py. However, when I exit out of python, the process crashes.  
```
Error in `python': corrupted size vs. prev_size: 0x000055beae3e0f00
======= Backtrace: =========
```

* **What is the expected behavior? What is the motivation / use case for changing the behavior?** report error with bad line code

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.** 
```
New Linecode.3P_OH_Code14 units=km nphases=3 Rmatrix=(1.41380960769992 0.05921682180272659 0.05921682180272659 0.05921682180272659 | 0.05921682180272659 1.41380960769992 0.05921682180272659 0.05921682180272659 | 0.05921682180272659 0.05921682180272659 0.9286412517222888 0.05921682180272659 | 0.05921682180272659 0.05921682180272659 0.05921682180272659 1.41380960769992 ) Xmatrix=(0.9898149352620426 0.48526905804901355 0.4936812762615934 0.48526905804901366 | 0.48526905804901355 0.9898149352620426 0.48526905804901355 0.44141980330426694 | 0.4936812762615934 0.48526905804901355 0.9547751759569331 0.48526905804901366 | 0.48526905804901366 0.44141980330426694 0.48526905804901366 0.9898149352620426 ) normamps=95.0 emergamps=0.0
```
What is wrong with this linecode is that the `nphases=3` indicates there are 3 conductors, but the `Rmatrix` has a 4*4 size, which expects there are 4 conductors, so there is a mismatch between these two attributes of a linecode.

### Support Question

PMeira, 2023-03-10:
Hi, thanks for reporting. This is an engine bug, like most reported here. I'm considering moving such issues to our hub repo (https://github.com/dss-extensions/dss-extensions) for visibility in the future.

I could reproduce with the upstream OpenDSS too, although the access violation exception there may happen down the line. This issue is probably inherited from very old code: there was already a check but that relied on Object Pascal exceptions, which probably worked fine on older Delphi versions and on 32-bit systems.

For the engine on DSS Extensions, per my local branch it now should error out with a message like:

> (#65534) Matrix Buffer in ParseAsSymMatrix too small. Check your input data, especially dimensions and number of phases.

Note that this still allows providing less elements than required (the unfilled elements are left as 0). We plan to add warnings someday, per https://github.com/dss-extensions/dss-extensions/issues/16

As soon as the new [DSS C-API](https://github.com/dss-extensions/dss_capi/) and DSS Python versions are out, a new version of OpenDSSDirect.py should be out with this fix, github.com/dss-extensions/OpenDSSDirect.py/issues/121 and misc OpenDSS updates. We'll probably release another version soon after with https://github.com/dss-extensions/OpenDSSDirect.py/pull/78 and other niceties.

PMeira, 2023-03-28:
Done and new packages uploaded to PyPI. We'll prepare a release message on GitHub soon too, for the moment see the relevant changes at https://github.com/dss-extensions/dss_python/blob/master/docs/changelog.md#0130

Sorry for the delay, some other issues need more time to investigate.

closed 2023-03-28 by PMeira

HHuuuhh666, 2024-12-26:
> 嗨，感谢您的报告。这是一个引擎错误，就像这里报告的大多数错误一样。我正在考虑将这些问题移至我们的中心存储库（https://github.com/dss-extensions/dss-extensions），以便将来查看。
> 
> 我也可以使用上游 OpenDSS 进行重现，尽管那里的访问冲突异常可能会在以后发生。此问题可能是从非常旧的代码继承而来的：已经有一个检查，但依赖于 Object Pascal 异常，这可能在较旧的 Delphi 版本和 32 位系统上运行良好。
> 
> 对于 DSS 扩展上的引擎，按照我的本地分支，它现在应该出现如下错误消息：
> 
> > (#65534) ParseAsSymMatrix 中的矩阵缓冲区太小。请检查您的输入数据，尤其是尺寸和相位数。
> 
> 请注意，这仍然允许提供少于所需数量的元素（未填充的元素保留为 0）。我们计划将来添加警告，具体见[dss-extensions/dss-extensions#16](https://github.com/dss-extensions/dss-extensions/issues/16)
> 
> 一旦新的[DSS C-API](https://github.com/dss-extensions/dss_capi/)和 DSS Python 版本发布，OpenDSSDirect.py 的新版本应该会发布，其中包含此修复、github.com [/ /issues/121](https://github.com/dss-extensions/OpenDSSDirect.py/issues/121)和其他 OpenDSS 更新。我们可能会在不久后发布另一个版本，其中包含#78和其他细节。
Clear
New Circuit.Circuit1 Phases=3 bus1=Bus1.1.2.3 basekv=220 pu=1.0 angle=0 MVAsc3=11297 MVAsc1=11297 x1r1=10 x0r0=10 
New Capacitor.Filter1 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye Kv=35 Cuf=93.37 XL=3.94 R=0 
New Capacitor.Filter2 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye Kv=35 Cuf=119.76 XL=1.72756 R=0 
New Capacitor.Filter3 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye Kv=35 Cuf=119.76 XL=1.10552 R=0 
New Line.Line1 bus1=Bus1.1.2.3 bus2=Bus2.1.2.3 Length=4.26 units=km R1=0.054 X1=0.308 C1=0 R0=0.054 X0=0.308 C0=0 
New LineCode.Line1 Nphases=3 Rmatrix=[0|0 0|0 0 0] Xmatrix=[0|0 0|0 0 0] 
New Load.Load1 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye vminpu=0.8 vmaxpu=1.2 Kv=35 kVA=155000 Pf=0.78 rneut=0 xneut=0 model=1 
New Load.Load2 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye vminpu=0.8 vmaxpu=1.2 Kv=35 kVA=26000 Pf=0.85 rneut=0 xneut=0 model=1 
New Load.Load3 Phases=3 bus1=Bus3.1.2.3.0 Conn=Wye vminpu=0.8 vmaxpu=1.2 Kv=35 Kw=0 kvar=180000 rneut=0 xneut=0 model=1 
New Capacitor.PCap1 Phases=1 bus1=Bus4.1.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[9.95 9.95 9.95] Cmatrix=[332.68|0 332.68|0 0 332.68] 
New Capacitor.PCap2 Phases=1 bus1=Bus5.2.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[9.95 9.95 9.95] Cmatrix=[332.68|0 332.68|0 0 332.68] 
New Capacitor.PCap3 Phases=1 bus1=Bus6.3.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[9.95 9.95 9.95] Cmatrix=[332.68|0 332.68|0 0 332.68] 
New Reactor.PRea1 Phases=1 bus1=Bus4.1.0 Conn=Wye Kv=20.2 Kvar=0 Rmatrix=[650|0 650|0 0 650] Xmatrix=[0|0 0|0 0 0] 
New Reactor.PRea2 Phases=1 bus1=Bus5.2.0 Conn=Wye Kv=20.2 Kvar=0 Rmatrix=[650|0 650|0 0 650] Xmatrix=[0|0 0|0 0 0] 
New Reactor.PRea3 Phases=1 bus1=Bus6.3.0 Conn=Wye Kv=20.2 Kvar=0 Rmatrix=[650|0 650|0 0 650] Xmatrix=[0|0 0|0 0 0] 
New Capacitor.SCap1 Phases=1 bus1=Bus3.1.0 bus2=Bus4.1.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[0 0 0] Cmatrix=[110.8933|0 110.8933|0 0 110.8933] 
New Capacitor.SCap2 Phases=1 bus1=Bus3.2.0 bus2=Bus5.2.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[0 0 0] Cmatrix=[110.8933|0 110.8933|0 0 110.8933] 
New Capacitor.SCap3 Phases=1 bus1=Bus3.3.0 bus2=Bus6.3.0 Conn=Wye Kv=20.2 Kvar=0 R=[0 0 0] XL=[0 0 0] Cmatrix=[110.8933|0 110.8933|0 0 110.8933] 
New Transformer.T2w1 Phases=3 XHL=13 %imag=0 %Noloadloss=0 Windings=2 Wdg=1 bus=Bus2.1.2.3.0 Conn=Wye Kv=220 Kva=160000 %R=0 Tap=1 rneut=0 xneut=0 Wdg=2 bus=Bus3.1.2.3 Conn=Delta Kv=35 Kva=160000 %R=0 Tap=1 
set DefaultBaseFrequency=50
set Basefrequency=50
set Frequency=50
set Algorithm=Newton
set Maxiter=50
set voltagebases=[220.0, 35.0, 20.2]
calcvoltagebases
solve
This is my DSS file. The result of executing opendss.exe using x64 and opendss.exe using x86 is different. Why is this?