dss-extensions/OpenDSSDirect.py #123: CITATION.bib

yasirroni, 2023-03-28:
### Feature Request

How about adding OpenDSSDirect.py a CITATION.bib?

```bib
@misc{OpenDSSDirect.py,
  author = {Dheepak Krishnamurthy and Paulo Meira and Max Zangs},
  title = {OpenDSSDirect.py},
  year = {2022},
  publisher = {GitHub},
  howpublished = {\url{https://github.com/dss-extensions/OpenDSSDirect.py}},
  note = {Accessed: 2023-03-27}
}
```

PMeira, 2023-03-28:
Good idea, I'm also adding a list of papers that cite or mention OpenDSSDirect.py and other DSS Extensions to the [main site](https://dss-extensions.org/) and [repo](https://github.com/dss-extensions/dss-extensions).

On the exact style of reference, let's wait a few days to see if someone has any suggestion. We could link to https://dss-extensions.org/OpenDSSDirect.py/ for example.

PMeira, 2023-06-13:
@yasirroni Thank you, I added the CFF as you suggested (probably easier to update it later), let's see if it works OK. Decided to go with the link to the website URL, by the way, but the repo it's fine too.

@kdheepak Remember to update your ORCID!

commit 23535552 (2023-06-13): Add CITATION.cff [skip ci]

Closes #123, closes #124.
closed 2023-06-13 by PMeira