dss-extensions/OpenDSSDirect.py #125: Generators - negative PF - reactive power

GSkrt, 2023-04-21:
### Versions

- [ ] Python version: 3.11.3
- [ ] Python architecture: 
- [ ] Operating system and version: Ubuntu
- [ ] OpenDSSDirect.py version number: 8,2

PMeira, 2023-04-21:
Could you provide a circuit to reproduce that?
It's unlikely to be a bug, but who knows...

GSkrt, 2023-04-21:
Not sure what is wrong. 

````
import opendssdirect  as dss
from opendssdirect import run_command
import os
import math

run_command("Compile " + os.path.relpath('test.dss'))

all_bus_names = dss.Circuit.AllBusNames()

cosfi_test=-0.95

brk = dss.Generators.First()
while brk > 0: 
    print(f"Before {dss.Generators.kW()}, , {dss.Generators.kvar()}")
    dss.CktElement.BusNames()[0]
    dss.Generators.PF(cosfi_test)  
    print(f"After {dss.Generators.kW()}, , {dss.Generators.kvar()}")
    brk = dss.Generators.Next()
    
  
dss.Solution.SolveSnap()
before = dss.Circuit.AllBusVMag()[::3]
    
brk = dss.Generators.First()
while brk > 0: 
    print(f"Before {dss.Generators.kW()}, , {dss.Generators.kvar()}")
    dss.CktElement.BusNames()[0]
    dss.Generators.kW(dss.Generators.kW())
    kvar = dss.Generators.kW() * math.tan(math.acos(cosfi_test))
    dss.Generators.kvar(kvar)
    print(f"After {dss.Generators.kW()}, , {dss.Generators.kvar()}")
    brk = dss.Generators.Next()
    
dss.Solution.SolveSnap()

after = dss.Circuit.AllBusVMag()[::3]
names = dss.Generators.AllNames()

for i in range(len(all_bus_names)): 
    if all_bus_names[i] == '741':
        print(f'{all_bus_names[i]} , {before[i]},  {after[i]}')
```

`

PMeira, 2023-04-21:
I meant the .dss script/circuit (this is not related to Python). Depending on how the circuit is defined, some properties may have different behaviors.

GSkrt, 2023-04-22:
```
Clear

! Note that this is a 3-wire delta system. 
! Node voltages in OpenDSS are normally line-ground and may give strange-looking results
! Be careful if you put line-neutral connected loads or other devices on this circuit.

New object=circuit.ieee37
~ basekv=230 pu=1.00 MVAsc3=200000 MVAsc1=210000

! Substation Transformer
New Transformer.SubXF Phases=3 Windings=2 Xhl=8
~ wdg=1 bus=sourcebus conn=Delta kv=230   kva=2500   %r=1
~ wdg=2 bus=799       conn=Delta kv=4.8   kva=2500   %r=1

! Load Transformer
New Transformer.XFM1  Phases=3 Windings=2 Xhl=1.81
~ wdg=1 bus=709       conn=Delta kv=4.80  kva=500    %r=0.045
~ wdg=2 bus=775       conn=Delta kv=0.48  kva=500    %r=0.045

! import line codes with phase impedance matrices
Redirect        linecodes.dss

! Lines
New Line.L1     Phases=3 Bus1=701.1.2.3  Bus2=702.1.2.3  LineCode=722  Length=0.96
New Line.L2     Phases=3 Bus1=702.1.2.3  Bus2=705.1.2.3  LineCode=724  Length=0.4
New Line.L3     Phases=3 Bus1=702.1.2.3  Bus2=713.1.2.3  LineCode=723  Length=0.36
New Line.L4     Phases=3 Bus1=702.1.2.3  Bus2=703.1.2.3  LineCode=722  Length=1.32
New Line.L5     Phases=3 Bus1=703.1.2.3  Bus2=727.1.2.3  LineCode=724  Length=0.24
New Line.L6     Phases=3 Bus1=703.1.2.3  Bus2=730.1.2.3  LineCode=723  Length=0.6
New Line.L7     Phases=3 Bus1=704.1.2.3  Bus2=714.1.2.3  LineCode=724  Length=0.08
New Line.L8     Phases=3 Bus1=704.1.2.3  Bus2=720.1.2.3  LineCode=723  Length=0.8
New Line.L9     Phases=3 Bus1=705.1.2.3  Bus2=742.1.2.3  LineCode=724  Length=0.32
New Line.L10    Phases=3 Bus1=705.1.2.3  Bus2=712.1.2.3  LineCode=724  Length=0.24
New Line.L11    Phases=3 Bus1=706.1.2.3  Bus2=725.1.2.3  LineCode=724  Length=0.28
New Line.L12    Phases=3 Bus1=707.1.2.3  Bus2=724.1.2.3  LineCode=724  Length=0.76
New Line.L13    Phases=3 Bus1=707.1.2.3  Bus2=722.1.2.3  LineCode=724  Length=0.12
New Line.L14    Phases=3 Bus1=708.1.2.3  Bus2=733.1.2.3  LineCode=723  Length=0.32
New Line.L15    Phases=3 Bus1=708.1.2.3  Bus2=732.1.2.3  LineCode=724  Length=0.32
New Line.L16    Phases=3 Bus1=709.1.2.3  Bus2=731.1.2.3  LineCode=723  Length=0.6
New Line.L17    Phases=3 Bus1=709.1.2.3  Bus2=708.1.2.3  LineCode=723  Length=0.32
New Line.L18    Phases=3 Bus1=710.1.2.3  Bus2=735.1.2.3  LineCode=724  Length=0.2
New Line.L19    Phases=3 Bus1=710.1.2.3  Bus2=736.1.2.3  LineCode=724  Length=1.28
New Line.L20    Phases=3 Bus1=711.1.2.3  Bus2=741.1.2.3  LineCode=723  Length=0.4
New Line.L21    Phases=3 Bus1=711.1.2.3  Bus2=740.1.2.3  LineCode=724  Length=0.2
New Line.L22    Phases=3 Bus1=713.1.2.3  Bus2=704.1.2.3  LineCode=723  Length=0.52
New Line.L23    Phases=3 Bus1=714.1.2.3  Bus2=718.1.2.3  LineCode=724  Length=0.52
New Line.L24    Phases=3 Bus1=720.1.2.3  Bus2=707.1.2.3  LineCode=724  Length=0.92
New Line.L25    Phases=3 Bus1=720.1.2.3  Bus2=706.1.2.3  LineCode=723  Length=0.6
New Line.L26    Phases=3 Bus1=727.1.2.3  Bus2=744.1.2.3  LineCode=723  Length=0.28
New Line.L27    Phases=3 Bus1=730.1.2.3  Bus2=709.1.2.3  LineCode=723  Length=0.2
New Line.L28    Phases=3 Bus1=733.1.2.3  Bus2=734.1.2.3  LineCode=723  Length=0.56
New Line.L29    Phases=3 Bus1=734.1.2.3  Bus2=737.1.2.3  LineCode=723  Length=0.64
New Line.L30    Phases=3 Bus1=734.1.2.3  Bus2=710.1.2.3  LineCode=724  Length=0.52
New Line.L31    Phases=3 Bus1=737.1.2.3  Bus2=738.1.2.3  LineCode=723  Length=0.4
New Line.L32    Phases=3 Bus1=738.1.2.3  Bus2=711.1.2.3  LineCode=723  Length=0.4
New Line.L33    Phases=3 Bus1=744.1.2.3  Bus2=728.1.2.3  LineCode=724  Length=0.2
New Line.L34    Phases=3 Bus1=744.1.2.3  Bus2=729.1.2.3  LineCode=724  Length=0.28
New Line.L35    Phases=3 Bus1=799r.1.2.3 Bus2=701.1.2.3  LineCode=721  Length=1.85

! Regulator - open delta with C leading, A lagging, base LDC setting is 1.5 + j3
new transformer.reg1a phases=1 windings=2 bank=reg1 buses=(799.1.2 799r.1.2) conns='delta delta' kvs="4.8 4.8" kvas="2000 2000" XHL=1
new regcontrol.creg1a transformer=reg1a winding=2 vreg=122 band=2 ptratio=40 ctprim=350 R=-0.201 X=3.348
new transformer.reg1c like=reg1a bank=reg1 buses=(799.3.2 799r.3.2)
new regcontrol.creg1c like=creg1a transformer=reg1c R=2.799 X=1.848
New Line.Jumper Phases=1 Bus1=799.2      Bus2=799r.2     r0=1e-3 r1=1e-3 x0=0 x1=0 c0=0 c1=0

! spot loads
New Load.S701a      Bus1=701.1.2 Phases=1 Conn=Delta Model=1 kV=  4.800 kW= 140.0 kVAR=  70.0
New Load.S701b      Bus1=701.2.3 Phases=1 Conn=Delta Model=1 kV=  4.800 kW= 140.0 kVAR=  70.0
[... 25 lines cut ...]
New Load.S742a      Bus1=742.1.2 Phases=1 Conn=Delta Model=2 kV=  4.800 kW=   8.0 kVAR=   4.0
New Load.S742b      Bus1=742.2.3 Phases=1 Conn=Delta Model=2 kV=  4.800 kW=  85.0 kVAR=  40.0
New Load.S744a      Bus1=744.1.2 Phases=1 Conn=Delta Model=1 kV=  4.800 kW=  42.0 kVAR=  21.0

New Generator.Test Bus=741 Phases=3 Kw = 500 kV = 4.8 Pf=1 Model=1 Vminpu=0.9 Vmaxpu=1.2 BaseFreq=60 enabled=True 

Set VoltageBases = "230,4.8,0.48"
CalcVoltageBases

! solve mode=direct
set maxiterations=100
solve

! show voltages LL Nodes
! show currents residual=y elements
! show powers kva elements
! show taps

```

I used this circuit as an example. I just added this line: 

`New Generator.Test Bus=741 Phases=3 Kw = 500 kV = 4.8 Pf=1 Model=1 Vminpu=0.9 Vmaxpu=1.2 BaseFreq=60 `

I work on a larger model based on GIS. Results are the same.

PMeira, 2023-04-23:
@GSkrt, that's perfect, thanks for the data.

By the way, if you ever need to share a circuit, running `obfuscate` should help to clean some private info, renaming circuit elements and buses (I think general objects like loadshapes and linecodes are left untouched). The `save circuit` command should work for most circuits, feel free to report if you find any issue (there are some issues in the official OpenDSS that we fixed for DSS Extensions already).

```python
dss.Text.Command('obfuscate')
dss.Text.Command('save circuit dir=some_dir_path')
```

> I think the first case should be negative too after `dss.Generators.PF(cosfi_test)` ?

If you solve the system after that, it is indeed be negative.

IEEE37 is a delta-delta system, so remember that when comparing the voltages. You could use the VLL function if it's more convenient.

Based on your example, I created the following. Note that I set the kW to 2000 to better observe the changes.

```python
from opendssdirect import dss
import numpy as np
from matplotlib import pyplot as plt

dss.Text.Command('redirect ../dss/electricdss-tst/Version8/Distrib/IEEETestCases/37Bus/ieee37.dss')
dss.Text.Command('New Generator.Test Bus=741 Phases=3 Kw=2000 kV=4.8 Pf=1 Model=1 Vminpu=0.9 Vmaxpu=1.2')

pf_deltas = np.linspace(-0.4, 0.4, 41)
pfs = np.sign(pf_deltas + 0.001) - pf_deltas
kvars = []
kVs = []
for pf in pfs:
    dss.Generators.First()
    dss.Generators.PF(pf)
    dss.Solution.SolveSnap()
    assert dss.Solution.Converged()
    dss.Circuit.SetActiveBus('741')
    volts = (np.asarray(dss.Bus.VLL()).view(dtype=complex))
    avg_volts = np.mean(abs(volts))
    kVs.append(avg_volts * 1e-3)
    kvars.append(dss.Generators.kvar())
    print(f"PF={pf:5g}, kw={dss.Generators.kW():4g}, kvar={dss.Generators.kvar():8g}, vavg={avg_volts:7g}")
    
plt.scatter(kvars, kVs, c=pfs, cmap='BrBG')
plt.colorbar(label='PF')
plt.xlabel('Reactive power (kvar)')
plt.ylabel('Vavg (kV)')
plt.show()
```

<details>
<summary>Text output</summary>

```
PF= -0.6, kw=2000, kvar=-2666.67, vavg=4167.04
PF=-0.62, kw=2000, kvar=-2530.97, vavg= 4210.3
PF=-0.64, kw=2000, kvar=-2401.17, vavg=4265.86
PF=-0.66, kw=2000, kvar=-2276.56, vavg=4315.41
PF=-0.68, kw=2000, kvar=-2156.51, vavg=4360.45
PF= -0.7, kw=2000, kvar=-2040.41, vavg=4402.34
PF=-0.72, kw=2000, kvar=-1927.71, vavg=4442.01
PF=-0.74, kw=2000, kvar=-1817.86, vavg=4480.11
PF=-0.76, kw=2000, kvar=-1710.32, vavg=4517.26
PF=-0.78, kw=2000, kvar=-1604.56, vavg=4560.94
PF= -0.8, kw=2000, kvar=   -1500, vavg=4599.86
PF=-0.82, kw=2000, kvar=-1396.01, vavg=4636.16
PF=-0.84, kw=2000, kvar=-1291.87, vavg=4672.62
PF=-0.86, kw=2000, kvar=-1186.73, vavg= 4712.7
PF=-0.88, kw=2000, kvar=-1079.49, vavg=4753.94
PF= -0.9, kw=2000, kvar=-968.644, vavg=4795.66
PF=-0.92, kw=2000, kvar=-851.996, vavg=4838.05
PF=-0.94, kw=2000, kvar=-725.903, vavg=4882.41
PF=-0.96, kw=2000, kvar=-583.333, vavg=   4931
PF=-0.98, kw=2000, kvar=-406.117, vavg=4989.35
PF=    1, kw=2000, kvar=       0, vavg=5074.96
PF= 0.98, kw=2000, kvar= 406.117, vavg=5123.34
PF= 0.96, kw=2000, kvar= 583.333, vavg=5131.61
PF= 0.94, kw=2000, kvar= 725.903, vavg=5143.38
PF= 0.92, kw=2000, kvar= 851.996, vavg=5150.42
PF=  0.9, kw=2000, kvar= 968.644, vavg=5154.18
PF= 0.88, kw=2000, kvar= 1079.49, vavg=5170.01
PF= 0.86, kw=2000, kvar= 1186.73, vavg=5184.86
PF= 0.84, kw=2000, kvar= 1291.87, vavg=5184.04
PF= 0.82, kw=2000, kvar= 1396.01, vavg=5197.33
PF=  0.8, kw=2000, kvar=    1500, vavg=5195.32
PF= 0.78, kw=2000, kvar= 1604.56, vavg=5211.05
PF= 0.76, kw=2000, kvar= 1710.32, vavg=5225.39
PF= 0.74, kw=2000, kvar= 1817.86, vavg=5222.96
PF= 0.72, kw=2000, kvar= 1927.71, vavg=5220.44
PF=  0.7, kw=2000, kvar= 2040.41, vavg=5235.75
PF= 0.68, kw=2000, kvar= 2156.51, vavg=5254.53
PF= 0.66, kw=2000, kvar= 2276.56, vavg= 5274.8
PF= 0.64, kw=2000, kvar= 2401.17, vavg=5274.91
PF= 0.62, kw=2000, kvar= 2530.97, vavg=5272.46
PF=  0.6, kw=2000, kvar= 2666.67, vavg=5296.18
```
</details>

Remember to update the path to `ieee37.dss`. 

From your initial post:

> Setting generators using dss.Generators.PF(-0.95) doasn't respond to lower voltage in distribution network.
>
> If you later check actual values dss.Generators.kvar() you get positive values for reactive power. Voltages in generator nodes should be lower but nothing happens.

The behavior seems correct. If you can confirm that, we can close this issue. I think you might have forgotten to solve after changing so the voltages were not updated.

Still, although it doesn't cause issues  with the solution, you indeed found a very minor bug.
When updating the PF, the internal logic first sets the kvar value, which is stored in two places. Right after that, it checks if the PF is negative and updates only one of those two values. The normal solution later updates the kvar from the correct value, so everything works as expected in the end, but if you check the "temporary" value before the solution, you'll get the wrong value.

I imagine there may be similar situations in other models, but since some require you to solve the system to get the updated values already, it's better to assume that you need to solve before getting the values for all models, including the generators.

Anyway, I already fixed [here](https://github.com/dss-extensions/dss_capi/commit/80c129e5537baaf38c3f8ce2f24a3fd3e2d7fc9b) (on DSS-Extensions) for our next version, and made a note to report back to the official OpenDSS in the next opportunity.  It's quite possible somebody already reported it years ago, but it's hard to find anything in the forum. Looking into the code history from the official SVN, the related code was introduced in 2009 (revision 177), so it's nothing new.

closed 2023-05-17 by PMeira