dss-extensions/OpenDSSDirect.py #129: How to export data to excel

Zzsheep1py, 2023-11-19:
### Versions

- [ ] Python version: 
- [ ] Python architecture: 
- [ ] Operating system and version: 
- [ ] OpenDSSDirect.py version number: 

### Feature Request

### Bug

* **What is the current behavior?**

* **What is the expected behavior? What is the motivation / use case for changing the behavior?**

* **What are the steps to reproduce this bug? Please provide a minimal working example of the bug if possible.**

* **Other information** (e.g. detailed explanation, stacktraces, related issues, suggestions how to fix, links for us to have context, eg. stackoverflow, gitter, etc)

### Support Question

PMeira, 2023-11-19:
There's no information in your question. Anything specific? 

Most csv/txt files created from the `export` can be opened in Excel directly.

You could also use various Python packages (e.g. openpyxl, xlsxwriter, pyexcel, or indirectly through pandas) to customize the output and create `.xlsx` files directly.

Zzsheep1py, 2023-11-19:
Thanks for your answer, I was wondering if there is a way to help me export the data from the simulation results directly to excel, such as LinesNames,from bus,to bus...

Zzsheep1py, 2023-11-19:
> There's no information in your question. Anything specific?
> 
> Most csv/txt files created from the `export` DSS command can be opened in Excel directly.
> 
> You could also use various Python packages (e.g. openpyxl, xlsxwriter, pyexcel, or indirectly through pandas) to customize the output and create `.xlsx` files directly.

Thanks for your answer, I was wondering if there is a way to help me export the data from the simulation results directly to excel, such as LinesNames,from bus,to bus...

PMeira, 2023-11-20:
> such as LinesNames,from bus,to bus...

Those are mostly input data. Depending on what you need, you can export to dataframes and use the `to_excel()` method to finally export to Excel files:

- `...utils...._to_dataframe`: there are dedicated functions for the DSS objects that have dedicated APIs, more at https://dss-extensions.org/OpenDSSDirect.py/notebooks/GettingStarted.html#Using-the-pandas-interface
- `...utils.class_to_dataframe`: this one will handle the DSS properties (as used in the .DSS scripts), see https://dss-extensions.org/OpenDSSDirect.py/notebooks/ActiveClass.html
- `ActiveClass.ToJSON`: this exports exports the DSS properties, but using a newer (work-in-progress) framework: https://dss-extensions.org/OpenDSSDirect.py/opendssdirect.html#opendssdirect.ActiveClass.ToJSON

A minimal sample for Lines below -- adjust the file path to point to a valid circuit script and give it a try to see which one you prefer. You probably already have pandas installed, but in case you don't, be sure to install it first.

```python
from opendssdirect import dss as odd, utils as odd_utils

odd.Text.Command('redirect IEEETestCases/13Bus/IEEE13Nodeckt.dss')

odd_utils.lines_to_dataframe().to_excel('Lines-lines_to_dataframe.xlsx')

odd_utils.class_to_dataframe('Line').to_excel('Lines-class_to_dataframe.xlsx')

import pandas as pd
odd.Basic.SetActiveClass('Line')
pd.read_json(odd.ActiveClass.ToJSON()).to_excel('Lines-from-json.xlsx')
```

We will also have a more uniform, performant and customizable method for exporting data, including both input circuit data and simulation state, very soon.

Zzsheep1py, 2023-11-20:
> > such as LinesNames,from bus,to bus...
> 
> Those are mostly input data. Depending on what you need, you can export to dataframes and use the `to_excel()` method to finally export to Excel files:
> 
> * `...utils...._to_dataframe`: there are dedicated functions for the DSS objects that have dedicated APIs, more at https://dss-extensions.org/OpenDSSDirect.py/notebooks/GettingStarted.html#Using-the-pandas-interface
> * `...utils.class_to_dataframe`: this one will handle the DSS properties (as used in the .DSS scripts), see https://dss-extensions.org/OpenDSSDirect.py/notebooks/ActiveClass.html
> * `ActiveClass.ToJSON`: this also exports the DSS properties, but using a newer (work-in-progress) framework: https://dss-extensions.org/OpenDSSDirect.py/opendssdirect.html#opendssdirect.ActiveClass.ToJSON
> 
> A minimal sample for Lines below -- adjust the file path to point to a valid circuit script and give it a try to see which one you prefer. You probably already have pandas installed, but in case you don't, be sure to install it first.
> 
> ```python
> from opendssdirect import dss as odd, utils as odd_utils
> 
> odd.Text.Command('redirect IEEETestCases/13Bus/IEEE13Nodeckt.dss')
> 
> odd_utils.lines_to_dataframe().to_excel('Lines-lines_to_dataframe.xlsx')
> 
> odd_utils.class_to_dataframe('Line').to_excel('Lines-class_to_dataframe.xlsx')
> 
> import pandas as pd
> odd.Basic.SetActiveClass('Line')
> pd.read_json(odd.ActiveClass.ToJSON()).to_excel('Lines-from-json.xlsx')
> ```
> 
> We will also have a more uniform, performant and customizable method for exporting data, including both input circuit data and simulation state, very soon.

Thank you very much, it's helpful

Zzsheep1py, 2023-11-28:
[image]

Can I use 'plot' to get this image and how should I get it?

PMeira, 2023-11-28:
> Can I use 'plot' to get this image and how should I get it?

Technically yes with some work. You could plot various circuits, accumulating the plots in a single matplotlib figure, changing line colors, etc.

Check the plotting examples at https://dss-extensions.org/dss_python/examples/Plotting/  
For more complex plots, see the last section, "Customizing using matplotlib’s API"

closed 2023-12-13 by PMeira